lagadic/visp #6: Issue with vpMath::round().

s-trinh, 2015-10-14:
The function vpMath::round should round a double to the nearest integer but there are some differences with the round function in math.h.
For example the code:
```
vpMath::round(5.5)
vpMath::round(-5.5)
```
produces 5 and -5 but should produces 6 and -6 ([example code](https://ideone.com/VldhPc)).

fspindle, 2015-10-14:
Fixed #6

closed 2015-10-14 by fspindle

s-trinh, 2015-10-16:
##### Complementary information:
In fact, there are multiple [rounding methods](https://en.wikipedia.org/wiki/Rounding) for floating point numbers.
The rounding mode for round function in math.h is [round half away from zero](http://en.cppreference.com/w/c/numeric/math/round) whereas the default rounding mode used in IEEE 754 computing functions and operators is [round half to even](https://en.wikipedia.org/wiki/Rounding#Round_half_to_even).
The difference can be seen [here](https://en.wikipedia.org/wiki/Rounding#Round_half_to_odd).