lagadic/visp #37: how to build for ios? build failed

hyuni, 2016-01-06:
follow this step [Tutorial: Installation from source on OSX for iOS devices](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-getting-started-iOS.html). 
but build failed

module visp_core : 
visp/modules/core/src/tools/file/vpIoTools.cpp:726:3: Call to unavailable function 'wordexp': not available on iOS

this my env setting :
CPU: 8-core 64-bit ivybridge
OS X: 10.10.5-x86_64
Xcode: 7.2
CLT: 7.2.0.0.1.1447826929
Clang: 7.0 build 700
X11: 2.7.8 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl

fspindle, 2016-01-06:
This was not tested a while !

Could you try to add in vpIoTools.cpp

#include <wordexp.h>

hyuni, 2016-01-06:
@fspindle thank you answer. but, already exist. apple deprecated wordexp functions.

fspindle, 2016-01-12:
A quick fix consists to comment the lines in vpIoTools.cpp:

    // wordexp_t exp_result;
    // wordexp(path.c_str(), &exp_result, 0);
    // path = std::string(exp_result.we_wordv[0]);
    // wordfree(&exp_result);

You can also detect if your platform is iOS and then disable the previous lines. This is done in the my commit that closes the issue.

closed 2016-01-12 by fspindle

hyuni, 2016-01-12:
@fspindle thanks, good day~~

commit 1b35daa1 (2016-01-12): Detect iOS/OSX platform and use wordexp() only if OSX (Closes: #37)
commit 7f79df83 (2016-01-12): Detect iOS/OSX platform and use wordexp() only if OSX (Closes: #37)