lagadic/visp #67: Encountered an error during the build process for iOS project

cop663, 2016-06-09:
After reading the official [guide](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-iOS.html), I was trying to build the ViSP library in Xcode for iOS project, here are the problems:

1. there is no target of "visp". I can only find "visp_XXX", such as "visp_core", "visp_io", etc.

2. Then I follow [this instruction](http://jaym2503.blogspot.my/2013/01/how-to-make-universal-static-library.html) to create a target named, "visp".

3. After adding the corresponding ".cpp" files to "Compile Sources" and ".h" files to "Copy Files" in Build Phases of "visp", I met the following errors:

build "visp"

[image]

build "ALL_BUILD"
[image]

Thank you!

fspindle, 2016-06-11:
The [tutorial](http://visp-doc.inria.fr/doxygen/visp-3.0.0/tutorial-install-iOS.html) is definitively out of date for ViSP 3.x. It needs an update. In the meantime, you will find some hints to proceed without using any 3rd parties:

Configure ViSP turning all 3rd parties, examples, tests... off:

	$ mkdir visp-build-ios; cd visp-build-ios
	$ ccmake ../visp -G Xcode -DCMAKE_INSTALL_PREFIX=./install -DBUILD_SHARED_LIBS=OFF -DUSE_OPENCV=OFF -DUSE_CPP11=OFF -DUSE_DC1394=OFF -DUSE_GSL=OFF -DUSE_JPEG=OFF -DUSE_LAPACK=OFF -DUSE_LIBUSB_1=OFF -DUSE_OPENCV=OFF -DUSE_PCL=OFF -DUSE_PNG=OFF -DUSE_PTHREAD=OFF -DUSE_REALSENSE=OFF -DUSE_X11=OFF -DUSE_XML2=OFF -DUSE_ZBAR=OFF -DUSE_ZLIB=OFF -DBUILD_TESTS=OFF -DBUILD_EXAMPLES=OFF -DBUILD_DEMOS=OFF -DBUILD_TUTORIALS=OFF

Check that there is no other 3rparty that is ON, turning all USE_<3rd party>=OFF, and proceed to the generation. Then open the Xcode project:

	$ open VISP.xcodeproj 

Click on ViSP in the left column to make appear configuration panel:
[image]

Then select the build settings
[image]

You have to change for each ``visp_<module>`` target the base SDK to Latest iOS. The following image shows how to do the things for ``visp_core``:
[image]

Do the same for  ``visp_detection``,  ``visp_gui``... ``visp_tt_mi``.
Now you may select your target (iPhone, iPad...)
[image]

You are now ready to build ViSP for an iPhone device using the play button
[image]

To finish the process, proceed to the installation selecting install target before applying play button
[image]

The build SDK containing includes and static libraries is available in ``./visp-build-ios/install``
 
PS: Since I haven't any iOS device under the hand, any feedback is welcome

cop663, 2016-06-11:
Thank you! fspindle 

There is still one problem after running the target of "install"

[image]

attached for your reference

Thank you!

technommy, 2016-06-11:
Got the same problem here, the library was successfully generated without any warning poped up, thanks so much : ) @fspindle

fspindle, 2016-06-11:
You haven't the rights to install in the default location ``/usr/local``. As given in my previous post, use ``ccmake -DCMAKE_INSTALL_PREFIX=./install ...`` to change the install path

fspindle, 2016-06-14:
I introduced a more elegant way to generate a ViSP framework for iOS. The tutorial is here http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-iOS.html

closed 2016-07-13 by fspindle