lagadic/visp #91: vpMeLine::computeRhoTheta should throw exception

mjcorriere, 2016-10-17:
In `vpMeLine::computeRhoTheta()` there is a condition that can be reached that causes a crash. There is a threshold difference value between two pixel intensities (v1 and v2) that sometimes is not reached. For example, `abs(v1-v2)` is never less than 1. When this occurs, the while loop continues, causing the indices `i1_, j1, i2_, j2_` to take on negative values. When they are then static cast to an `unsigned int`, these indices become astronomically large ints, which cause out of range exceptions when attempting to access the image `I[i1_][j2_]`.

See the code below.

```c++
    while (!end)
    {
      end = true;
      unsigned int i1_ = static_cast<unsigned int>(i1);
      unsigned int j1_ = static_cast<unsigned int>(j1);
      unsigned int i2_ = static_cast<unsigned int>(i2);
      unsigned int j2_ = static_cast<unsigned int>(j2);
      v1=I[i1_][j1_];
      v2=I[i2_][j2_];
      if (abs(v1-v2) < 1)
      {
        incr-- ;
        end = false ;
        if (incr==1)
        {
         /*****************************************/
         /* SHOULD THROW EXCEPTION OR QUIT THE LOOP HERE */
         /*****************************************/
          std::cout << "In CStraightLine::GetParameters() " ;
          std::cout << " Error Tracking " << abs(v1-v2) << std::endl ;
        }
      }
      update_indices(theta,i,j,incr,i1,i2,j1,j2);
    }
```

fspindle, 2016-10-18:
Could you attach an image that causes the crash

mjcorriere, 2016-10-21:
Unfortunately for work reasons I cannot. But, I can tell you it is a grayscale image, overexposed near the region where the line is being tracked. I believe what happens is the center of the line drifts over a fully overexposed region, and samples nothing but 255 values in both directions. As such, it never is able to find a difference between v1 and v2.

fspindle, 2016-10-21:
Ok, I see. I will check that asap.

commit 8a131f71 (2016-10-28): In me line tracker fix a possible infinite loop during rho sign computation when there is no gray level difference between both line sides. (Closes #91)
closed 2016-10-28 by fspindle