lagadic/visp #92: Is there any interface to resize display window?

mizulily, 2016-10-18:
I read image stream from a Basler camera with image size 1600\*1200. Since my screen resolution is 1440*900, the image is tiled on the screen and not complete. I can't initialize a tracker by clicks very well, because not all parts of the image can be seen at the same time.
If I use vpDisplayX method, the window can be adjusted but the image is still that big size. If I use vpDisplayOpenCV method, the default window can not be adjusted. I tried to look up at the source code of vpDisplayOpenCV. The variable to create opencv window is CV_AUTOSIZE, so the window is strictly the same size of the image.
I started to use ViSP very recently, so I am not sure if there is already an existing interface to resize the display window size, or this function is not yet supported?

fspindle, 2016-10-18:
Display rescaling feature is not implemented in ViSP.

1/ You can downscale the image using `vpImage::halfSizeImage()` and work only with the smaller image. In that case, if used camera parameters should be also adapted. 

2/ You can also use the original image for image processing and create a downscaled image that is only used for display capabilities (ie. user interaction using `getClick()`). The pixel coordinates returned by `getClick()`should then be adapted to the original size image (ie. multiplied by 2).

fspindle, 2016-12-03:
I have introduced this feature that was also requested by other users. See http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-image-display.html

closed 2016-12-03 by fspindle