lagadic/visp #104: vpMomentAlpha: Results are not repeatable

xoox, 2016-12-06:
Hello, the example code was modified to test with an image.
But when running the program several times, the results were not the same.
Is that a bug?

Here is the results:

```
$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = 0.0315403rad = 1.80712deg

mu3=-3.14023e-05        0.000310706     2.21268e-06     0.000537459

$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = 0.0315403rad = 1.80712deg

mu3=-3.14023e-05        0.000310706     2.21268e-06     0.000537459

$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = -0.00434398rad = -0.248892deg

mu3=-0.000185569        2.89679e-05     -3.46395e-05    -0.000129675

$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = 0.0315403rad = 1.80712deg

mu3=-3.14023e-05        0.000310706     2.21268e-06     0.000537459

$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = -0.0227243rad = -1.30201deg

mu3=-0.000290305        -0.000161996    -5.60761e-05    -0.000563486
```

We can see the results of 1st, 2nd and 4th run are identical. But results of 3rd and 5th run have different values.

The code and testing files are attached.

Our testing platform is Ubuntu 12.04.5 LTS
kernel: 3.14.17-xenomai-2.6.4 #1 SMP Thu Nov 10 20:42:27 CST 2016 i686 i686 i386 GNU/Linux

xoox, 2016-12-06:
To get the gravity center with a line
```cpp
std::cout << "Gravity Center:" << gRef.getXg() << gRef.getYg() << std::endl;
```

Different results would also be gotten when run more than 10 times.
```
$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = 0.0315403rad = 1.80712deg

mu3=-3.14023e-05        0.000310706     2.21268e-06     0.000537459
Gravity Center:0.0132962, 0.0730665

$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = -0.0227335rad = -1.30253deg

mu3=-0.000290363        -0.000162101    -5.60871e-05    -0.000563722
Gravity Center:0.0188937, 0.103826
```

manisoftwartist, 2016-12-06:
@xoox : +1 for providing the test code and test image. Was easy to narrow down the problem. 
`objRef.fromImage(I, 127, cam);`  was the API used with binary images. Perhaps, a few bits get toggled while doing the thresholding and hence the difference in values. I replaced this line with `objRef.fromImage(I, cam, vpMomentObject::BLACK);` which is for grayscale images. For your test image, I get consistently the same values below for 10 consequent runs. Could you please check and confirm us if this fixed this for you?
'''
Order of vpMomentObject: 3
m[0,0] = 0.379687	m[1,0] = 0.00502855	m[2,0] = 0.015774	m[3,0] = 0.000596269	
m[0,1] = 0.0299691	m[1,1] = 0.000455113	m[2,1] = 0.00146488	
m[0,2] = 0.0164127	m[1,2] = 0.000226118	
m[0,3] = 0.00391425	

--- Reference object ---
alphaRef=/user/mbakthav/home/Work/code/visualservo/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = 0.0350007rad = 2.00539deg 

mu3=-2.86967e-05	0.000218284	-4.38618e-07	0.000401273	
'''

xoox, 2016-12-07:
Thanks for your timely response @manisoftwartist
`objRef.fromImage(I, cam, vpMomentObject::BLACK)` do get same results.

Have you gotten different results with my original code?

Actually I tested with binary image in the beginning.
Replace
```cpp
objRef.fromImage(I, 127, cam);
``` 
with
```cpp
    vpImage<unsigned char> Icanny;
    vpImageFilter::canny(I, Icanny, 5, 65, 3);
    objRef.fromImage(Icanny, 127, cam);
```
I got different results with just 2 consequent runs.
```
$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = -1.54452rad = -88.4945deg

mu3=4.4507e-07  -6.51162e-06    8.45356e-07     -1.62884e-06
Gravity Center:0.00932785, 0.0865824

$ ../build/bin/test_tmvMoments -c camera.xml -r p_080000290_1.jpg
Found camera with name: "Camera"
--- Reference object ---
alphaRef=/home/cai/visp/modules/core/src/tracking/moments/vpMomentAlpha.cpp
Alpha = 1.38612rad = 79.4186deg

mu3=1.91282e-05 -1.00657e-05    9.48664e-06     -4.13108e-05
Gravity Center:-0.0450212, 0.136993
```

manisoftwartist, 2016-12-07:
>Have you gotten different results with my original code?
   Yes, the results were different.
> Actually I tested with binary image at the beginning.
   Even with binary images, I am afraid some pixels are toggled between 0/1 when they have 127 for 
   example (unverified untested hypothesis)

You could print out the values of the raw moments first. You can use
`vpMomentObject::printWithIndices(objRef, std::cout);` and compare your results between the runs.

fspindle, 2016-12-07:
I was able to reproduce the issue on Ubuntu 16.04 (but not on OSX !)

When I disable openmp usage during ViSP build (-DUSE_OPENMP=OFF) and run again the test, I got always the same values with the original test code.

So it seems that there is something going wrong in https://github.com/lagadic/visp/blob/master/modules/core/src/tracking/moments/vpMomentObject.cpp#L291 when using openmp.

fspindle, 2016-12-07:
Removing the line https://github.com/lagadic/visp/blob/master/modules/core/src/tracking/moments/vpMomentObject.cpp#L291 seems enough to fix this issue:

    //#pragma omp parallel shared(threshold)

@xoox Could you test on your side ?

xoox, 2016-12-07:
Thanks @fspindle
I'll test it tomorrow morning UTC+8
BTW, removing `#pragma omp parallel shared(threshold)` directives seems to lost openmp performance boost on SMP platforms.
A better fix is still needed.

I remember having encountered similar non-repeatable issues with 2D feature detection with VISP or OpenCV. I'll renew reports if similar issues are replayed with other modules.

commit 73a37fce (2016-12-07): Reintroduce #pragma omp atomic to fix non repeatable moments computation (Closes #104)
closed 2016-12-07 by fspindle