lagadic/visp #105: vpMomentAlpha: Question about symmetric judgement based on centered moments

xoox, 2016-12-06:
in [modules/core/src/tracking/moments/vpMomentAlpha.cpp](https://github.com/lagadic/visp/blob/master/modules/core/src/tracking/moments/vpMomentAlpha.cpp)
line 62
```cpp
for (std::vector<double>::iterator it = ref_.begin(); it!=ref_.end(); ++it)
    if (*it<=1e-4)
      symmetric = true;
```
Since moments are negative sometimes, is it right Line 63 `if (*it<=1e-4)` be changed to `if (abs(*it)<=1e-4)`?

manisoftwartist, 2016-12-06:
Perhaps. If you created this suspecting the problem in #104, it has no relation. I have verified.

xoox, 2016-12-07:
Right. This issue is not related to #104 
I'm not familiar with theories, But this line of code seems not be the right way.

fspindle, 2016-12-07:
[Line 63](https://github.com/lagadic/visp/blob/master/modules/core/src/tracking/moments/vpMomentAlpha.cpp#L63) should be as:

    if (std::fabs(*it)<=1e-4)

and [Line 56](https://github.com/lagadic/visp/blob/master/modules/core/src/tracking/moments/vpMomentAlpha.cpp#L56) should be 

    \f$\mu_{30},\mu_{21},\mu_{12},\mu_{03}\f$

@manisoftwartist Could you confirm ? Thanks

manisoftwartist, 2016-12-07:
Yes. seems fine for std::abs(). In line 56, the order of the moments should not matter.

commit 5413c177 (2016-12-07): Fix vpMomentAlpha symetric detection (Closes #105)
closed 2016-12-07 by fspindle