lagadic/visp #109: Link errors with undefined reference to TIFF

fspindle, 2016-12-08:
System: Ubuntu 16.04 LTS
Any version of ViSP

During ViSP build you may encounter the following build issues:
````
$ cd visp-build
$ cmake ../visp
$ make
...
[ 12%] Linking CXX executable testIoPPM
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFOpen@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFGetField@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFClose@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4.9: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
````

fspindle, 2016-12-08:
These errors may appear after installation of [anaconda](https://www.continuum.io/downloads#linux).

If you follow the default anaconda installation, you can see that anaconda has a set of libraries in `~/anaconda2/lib` like `libtiff.so`, `libjpeg.so`,  `libxml2.so` ...  that are also installed in `/usr/lib` but with other versions. The result is that the corresponding system libraries in `/usr/lib/x86_64-linux-gnu` may be hidden by files in `~/anaconda2/lib`.

The solution I found to be able to build ViSP with OpenCV is to remove `anaconda2/bin` folder from `PATH`.

````
$ diff ~/.bashrc ~/.bashrc-anaconda2.bak
125,127d124
<
< # added by Anaconda2 4.2.0 installer
< export PATH="/home/user/anaconda2/bin:$PATH"
````
editing `~/.bashrc` and commenting

````
#export PATH="/home/user/anaconda2/bin:$PATH"
````

Then you need to open a new terminal in order to have a proper `PATH` var without `~/anaconda2/bin` additional folder.

In this new terminal do a fresh build of ViSP with all default 3rd party libraries including OpenCV
````
$ cd visp-build
$ rm -rf *
$ cmake ../visp
$ make -j4
````

closed 2016-12-08 by fspindle

jaewoosong, 2017-04-12:
@fspindle 's comment works!

doem97, 2020-10-11:
@fspindle's method surely will work - but not necessarily remove all the anaconda path manually from the $LD_LIBRARY_PATH or $PATH. If you don't want Anaconda libs be included into the system $PATH or $LD_LIBRARY_PATH, just `conda deactivate`. This command will clean the changes anaconda have done to the $PATH. Your can check with `export` that only anaconda/condabin in $PATH, which is okay.
Make sure when you use cmake your path is clean with only anaconda/condabin included.