lagadic/visp #139: Create a tutorial for basic drawings

fspindle, 2017-03-06:
This tutorial should show how to use the [vpDisplay](http://visp-doc.inria.fr/doxygen/visp-daily/classvpDisplay.html) class to draw in overlay a point, a cross, a segment, a circle, a rectangle modify the thickness of the drawings...

ashwinpathak20, 2018-01-04:
Hi @fspindle, I would like to work on this issue if it is still available, as I am new to this organization and would like to contribute and learn more about ViSP.

fspindle, 2018-01-05:
Feel free to contribute and propose a PR

ashwinpathak20, 2018-01-05:
Do I just need to write a c++ program for it or make report explaining it
too?

fspindle, 2018-01-05:
I see a tutorial `doc/tutorial/gui/tutorial-gui.doc` named `Tutorial: GUI module overview` in a new ["GUI module" subsection](https://github.com/lagadic/visp/blob/master/doc/mainpage.doc.in#L99) that explains that there are different classes `vpDisplayX`, `vpDisplayGDI`, `vpDisplayD3D`, `vpDisplayGTK`, `vpDisplayOpenCV` that inherit from `vpDisplay` that could be used to:
1. display an image in a windows
2. make drawings (lines, circle, rectangle, polyline)  in the window overlay
3. add text in the window overlay
4. export the image and all the drawings (geometric forms + text) in a new image using `vpDisplay::getImage()`
5. handle keyboard events
6. handle mouse events

These classes are part of the [`GUI` module](https://visp.inria.fr/gui/) and are using [3rd parties](https://visp.inria.fr/3rdparty_gui/). Depending on the OS the user may use one of these classes. Why not introducing a table that resumes this (classes versus OS).

Each capability (1. to 6.) could be a section of the tutorial. Each section should be illustrated with a `.cpp` example that could be introduced in `tutorial/gui`. There is already `tutorial/image/tutorial-viewer.cpp` that could be reused to illustrate 1. If there is the need to use an image as input, you may use `monkey.ppm` color image that is in the `tutorial/image` folder and that should be copied in the new `tutorial/gui`folder. The [`tutorial/image/CMakeFiles.txt`](https://github.com/lagadic/visp/blob/master/tutorial/image/CMakeLists.txt) file should be used as an example to copy then the `tutorial/gui/monkey.ppm` file to the right build folder.

Don't hesitate to propose other things...

ashwinpathak20, 2018-01-17:
@fspindle , I have written all the codes corresponding to each sections. I am in the process of writing tutorial, however, I am not able to figure out, how to add snippets in the .doc file for tutorials. Please guide me through.

Thanks

fspindle, 2018-01-17:
[Here](https://github.com/lagadic/visp/blob/master/tutorial/tracking/blob/tutorial-blob-auto-tracker.cpp#L28) is an example where line 28 is added as a snippets in [tutorial-tracking-blob.doc](https://github.com/lagadic/visp/blob/master/doc/tutorial/tracking/tutorial-tracking-blob.doc#L40)

See also doxygen [manual](https://www.stack.nl/~dimitri/doxygen/manual/commands.html#cmdsnippet)

fspindle, 2019-01-06:
Done with PR #489

closed 2019-01-06 by fspindle