lagadic/visp #140: Create a tutorial that shows how to load, display and modify an image

fspindle, 2017-03-06:
The image modification could be just the conversion to a gray level image from a color image. Other ideas are also welcome.

sudoankit, 2018-02-15:
Visp is pretty interesting! I'll work on this to start my contributions!

yshkp, 2018-12-10:
Working on it and planning for the modification of image by distinctly adjusting its RGB values. Any ideas regarding this how to process ?

yshkp, 2018-12-10:
> There is already this PR #300 <https://github.com/lagadic/visp/pull/300>
> that was never finished by it's author.
> You can start from it.

Yes Sure, will start working on it and I am also looking to contribute in a
project through GSOC 2019 as i have keen interest towards Computer Vision
and VIsp offers really easy and impressive platform to nourish skills and
perform at higher level.

fspindle, 2019-01-07:
Today there are already these [image processing tutorials](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial_imgproc.html) that address this enhancement.

closed 2019-01-07 by fspindle