lagadic/visp #143: Create a tutorial using ViSP and Matlab

fspindle, 2017-03-06:
It should show how to create a C++ project with CMake that mix ViSP and Matlab

redgene, 2018-03-05:
Hi Fabien,
I find this tutorial intersecting with my skills in Matlab. I would like to know the following:
1. Do you want to call Matlab code from ViSP? 
2. Is the Matlab code pre-compiled?

Thanks,
Utkarsh

fspindle, 2018-03-05:
Answer to 1. is yes
Answer to 2. is no

A good starting would be in a `main()` to open Matlab engine, create a `vpMatrix x`, pass this matrix that is row major to Matlab (that is col major), apply a Matlab function `[y] = func(x)` to the matrix and get the results from Matlab engine.

redgene, 2018-03-07:
Hi Fabien,
I have completed the tutorial which calls a MATLAB function from ViSP program using MATLAB engine and have created a repository [ViSP_MATLAB](https://github.com/redgene/ViSP_MATLAB). Please let me know if I need to create a pull request for this or if this needs to be improved.

In this basic tutorial, I do the following:
1. Create and Open an instance of MATLAB engine, 
2. Copy data from a vpMatrix object 'x' to a MATLAB mxArray variable 'T',
3. Put the variable T in MATLAB workspace,
4. Compute column-wise sum of the variable by evaluating MATLAB expression, taking into account the column-major format of MATLAB,
5. Get the variable D from MATLAB workspace,
6. Finally copy the contents of D to local output variable (Please see Result.jpg)

CMakeLists.txt consists of the following MATLAB R2017b include directory(MATLAB_INCLUDES), library directory (MATLAB_LIB_PATHS) and static libraries ( MATLAB_LIBS: libeng.lib & libmx.lib). Please set them according to your MATLAB installation:

```
project(VISP_MATLAB)

cmake_minimum_required(VERSION 2.6)

find_package(VISP REQUIRED visp_core)
include_directories(${VISP_INCLUDE_DIRS})

set(MATLAB_LIBS C:/Program\ Files/MATLAB/R2017b/extern/lib/win64/microsoft/libeng.lib C:/Program\ Files/MATLAB/R2017b/extern/lib/win64/microsoft/libmx.lib)
set(MATLAB_INCLUDES C:/Program\ Files/MATLAB/R2017b/extern/include)
set(MATLAB_LIB_PATHS C:/Program\ Files/MATLAB/R2017b/extern/lib/win64/microsoft)
include_directories(${MATLAB_INCLUDES})
link_directories(${MATLAB_LIB_PATHS})

add_executable(main main.cpp)
target_link_libraries(main ${VISP_LIBRARIES} ${MATLAB_LIBS})

```
Thanks,
Utkarsh

fspindle, 2018-03-30:
Hi Utkarsh,

Thanks for your contrib.
Your `CMakeLists.txt` file is too much hardcoded. You should modify to:
````
project(VISP_MATLAB)

cmake_minimum_required(VERSION 2.6)

find_package(VISP REQUIRED visp_core)
include_directories(${VISP_INCLUDE_DIRS})

# Matlab
find_package(Matlab COMPONENTS MX_LIBRARY ENG_LIBRARY MAT_LIBRARY REQUIRED)
include_directories(${Matlab_INCLUDE_DIRS})
add_executable(main main.cpp)
target_link_libraries(main ${VISP_LIBRARIES} ${Matlab_LIBRARIES})
````

I haven't time to check if all the components `MX_LIBRARY ENG_LIBRARY MAT_LIBRARY ` are requested. You can test.

What I see now is a PR to ViSP with the code (`README`, `CMakeLists.txt`, `main.cpp`) in `visp\tutorial\matlab` and a nice tutorial in `tutorial\doc\matlab\matlab.doc` that gives some explanations, maybe an image with the result. It should explain that Matlab is found if the path to Matlab binary is in the PATH environment var, or if `Matlab_ROOT_DIR` environement var is set. Writing this tutorial will imply some modifications in  `visp\tutorial\tutorial.doc` and  `visp\tutorial\mainpagel.doc.in`.

fspindle, 2018-04-01:
Following the contributing [page](https://visp.inria.fr/contributing-code/) could you please also fill the right transfer [form](http://visp-doc.inria.fr/right-transfert/visp-right-transfer.pdf)

redgene, 2018-04-02:
I am working on creating the tutorial document and will create a PR as soon as it is done. 
And I will send the Rights Transfer form.

closed 2018-04-13 by fspindle