lagadic/visp #144: Create a tutorial saving images from a stream

fspindle, 2017-03-06:
It should explain how to save an image sequence from a video stream acquired by a camera.

yurriy, 2017-03-17:
I'll do it

fspindle, 2017-03-17:
Welcome

yurriy, 2017-03-17:
@fspindle,
there already exists tutorial-video-recorder.cpp. Should the tutorial just explain it, or there must be something new?

fspindle, 2017-03-18:
This new tutorial could be referenced in [Tutorial: Image frame grabbing](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-grabber.html) in the [next section](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-grabber.html#grabber-next).

It should explain tutorial-video-recorder.cpp that is a good starting and any else other useful things that you think that could be useful for beginners.

Feel also free to add new cpp if you think it could help.

yshkp, 2019-01-06:
@fspindle  Section already includes its explanation. Do you want brief of code working ? Then I can look into it.

fspindle, 2019-01-06:
@yshkp The [Tutorial: Image frame grabbing](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-grabber.html) already answer this enhancement. I will close it.

closed 2019-01-06 by fspindle