lagadic/visp #145: Improve video reader to determine the first and last image of a sequence of images

fspindle, 2017-03-06:
The [video reader class](http://visp-doc.inria.fr/doxygen/visp-daily/classvpVideoReader.html) should be improved to determine the first and last image of a sequence efficently.

Let us consider a folder with images ``image%04d.png`` where the first image is ``image2000.png`` and the last ``image2345.png`` the class should be able to determine that the first image has number 2000 and the last 2345.

shaunakdixit, 2017-03-16:
sir
I would like to work upon this 
since I had gone through all the tutorials.
Maybe I can solve this one.

yurriy, 2017-03-17:
Hi, @shaunakdixit
Do you want to do it for Windows and Unix both? I can make OS-dependent part, if you're not against.

shaunakdixit, 2017-03-18:
Hi, @yurriy sure you can go forward.
Although due to exams I can not work on it for 2-3 days.

shaunakdixit, 2017-03-19:
hi, @fspindle 
Sir, I was implementing a new function of class vpVideoReader in an example.
but it is showing an error "has no member new_function()".
although I have built the visp again after implementing the new function in vpVideoReader.h and vpVideoReader.cpp

fspindle, 2019-01-07:
Already done

closed 2019-01-07 by fspindle