lagadic/visp #146: Introduce geometric transformation of images

fspindle, 2017-03-06:
Today there is noting to rotate or translate an image. Could be useful to introduce these features.

vikasTmz, 2017-03-12:
Hello @fspindle , I'd like to give this a shot. I've written codes on Affine transformation before so would like to implement this and also understand the code base in the process.

fspindle, 2017-03-13:
Ok, you are welcome.

vikasTmz, 2017-03-14:
Can you give me more details on this enhancement?

fspindle, 2017-03-14:
What I see, but could be discussed is
- output image has the same size than input image
- translation: from a translation vector with 2 doubles as coordinates
- rotation: set the center of rotation (2 double) and the angle in radian
- a scaling factor could also be considered

fspindle, 2017-03-14:
I better idea would be to implement a more generic function (warpAffine) that takes an affine transformation as input and where each pixel u,v is warped in u_w, v_w using the formula:

u_w = c_u + a_1 * u + a_2*v
v_w = c_v + a_3 * u + a_4*v

In case of a pure translation a_1 = a_2 = a_3 = a_4 = 0
in case of rotation with origin the middle of the image c_u = width/2 and c_v = height/2

vikasTmz, 2017-03-14:
So, `warpAffine` would take transformation matrix **T** as input?
> u_w = c_u + a_1 * u + a_2v
v_w = c_v + a_3 * u + a_4v

=> **T** = [ a_1 a_3 0 ; 
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; a_2 a_4 0 ; 
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;  c_u c_v 1 ]

I've already implemented a function (_in matlab_) which uses backward transformation (inverse mapping) to warp images, so I know the theory behind this.

Where should this be implemented i.e under which class will `warpAffine` belong to?

fspindle, 2017-03-15:
``vpImageTools`` class would be a good candicate

fspindle, 2019-01-07:
Continued in #491

closed 2019-01-07 by fspindle