lagadic/visp #154: Improve vpVideoReader to be able to get first and last frame without calling open()

fspindle, 2017-03-20:
In PR #151 getFirstFrame() and getLastFrame were improved, but these methods can only be used after a call to open() like in the following code:

````
#include <visp3/visp.h>

int main(int argc, const char *argv[])
{
  vpImage<unsigned char> I;
  vpVideoReader grabber;
  grabber.setFileName(argv[1]);
  grabber.open(I);
  std::cout << "First: " << grabber.getFirstFrameIndex() << " last: " << grabber.getLastFrameIndex() << std::endl;
}
````

A nice improvement would be to be able to get first and last frame number of a sequence of images without calling open() like:

````
#include <visp3/visp.h>

int main(int argc, const char *argv[])
{
  vpVideoReader grabber;
  grabber.setFileName(argv[1]);
  std::cout << "First: " << grabber.getFirstFrameIndex() << " last: " << grabber.getLastFrameIndex() << std::endl;
}
````

yurriy, 2017-03-21:
I'll do it

yurriy, 2017-03-31:
It must work only for image sequences?

fspindle, 2017-03-31:
Good question. 
In this enhancement we can consider only image sequences.

PR #215 (2017-08-25): Update video properties (first and last frame, framerate) without ope…
…n() call; closes #154 

This PR replace PR #186 

Migrate major content of open() into getProperties()
commit d077f3b0 (2017-08-25): Update video properties (first and last frame, framerate) without open() call; closes #154
closed 2017-08-25 by fspindle