lagadic/visp #155: Improve vpVideoReader to introduce a frame step

fspindle, 2017-03-20:
vpVideoReader class uses a constant frame step equal to 1, meaning that the next frame number to open is equal to the current frame number + 1.

It would be nice to introduce a new function ``setFrameStep()`` that allows the user to set the frame step. 

Up to now, vpVideoReader cannot handle the following sequence of images: ``image10.png``, ``image20.png``, ``image30.png``. Introducing ``setFrameStep(10)`` could solve this issue.

As done in #151 it could be nice to go further and determine automatically the frame step of a sequence to be able to handle automatically a sequence without calling ``setFrameStep()``

Lrakulka, 2017-03-20:
I want to work on it.

fspindle, 2017-03-20:
Welcome

Lrakulka, 2017-03-25:
I have a question.
What should return the method `findLastFrameIndex()` after my improvements?
Example:
We have a sequence of frames: 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16
We set `setFrameStep()` to 5
We call the method `findLastFrameIndex()`
What should return the method? (last index = 15 or 16)

fspindle, 2017-03-26:
`findLastFrameIndex()` don't have to be changed in this enhancement since it will be the result of PR #151 .
To answer your question `findLastFrameIndex()` will return 16.

Lrakulka, 2017-04-01:
#167

PR #167 (2017-05-19): Issue#155 set the frame step
Hello,

I marked the important changes that I made with the comments "an important change".
I did it because I changed the code to the ViSp's standard, and it is difficult to understand what changes were made.

If all is okay, I will do automatic detection of the frame step for an image sequence.

fix #155  issue
closed 2017-05-19 by fspindle