lagadic/visp #178: initFromPoints

wangbf, 2017-05-04:
I used the vpMbTracker::initFromPoints method instead of the initialise manually the pose by clicking on the image, here is my init file:
4                  # Number of points
0     0      0     # Point 0
0.165 0      0     # Point 3
0.165 0     -0.08  # Point 2
0.165 0.068 -0.08  # Point 5
4 
84	194    
208	350   
375	330
348	460  

After run, I got "Points are collinear" error

fspindle, 2017-05-04:
Your 3D points are not collinear, that's why I can't understand the error. 
Moreover I'm not able to reproduce the issue with your data.

To reproduce the issue, could you send me an image, the cad model, the xml file that are used...

wangbf, 2017-05-05:
I use the "tutorial-mb-edge-tracker.cpp",  the cad model is the teabox.cao, the xml file is the teabox.xml, I did't modity them, just modify the teabox.init, add the flow lines:
4
84	194
208	350
375	330
348	460

then, I use **tracker.initFromPoints(I, objectname + ".init")** to instead of **the tracker.initClick(I, objectname + ".init", true)**, like this:
[image]

I got erro in my console:
[image]

when I change back, use **the tracker.initClick(I, objectname + ".init", true)**, it work.

fspindle, 2017-05-05:
The issue comes from the comments `# bla bla` that are not managed in `initFromPoint()`. I will do the changes in the code. In the mean time you may remove the comments in the file

commit cf7ff489 (2017-05-05): Handle potential comments in the mbt init file used to initialize 2D and 3D point coordinates (Closes #178)
closed 2017-05-05 by fspindle