lagadic/visp #190: Matrix multiplication enhancement suggestion

amagasso, 2017-06-26:
Hi,
Maybe for large matrix we could consider using a BLAS library like Lapack to perform matrices multiplication?
Of course, we will have to test if the performance is improved or not.

fspindle, 2017-06-27:
In your use case, what is the size of your large matrices ?

s-trinh, 2017-06-27:
In my opinion, with interaction matrix like size (`Nx6`), the current implementation could be enough.
But for general use and with large matrix, [BLAS library](https://stackoverflow.com/questions/1303182/how-does-blas-get-such-extreme-performance) should be more efficient.
One issue I see is maybe the overhead time to convert back and forth the data matrix, if any.
This could be a possible future enhancement (validated by proper benchmarks).

PR #192 (2017-06-29): Add Lapack support for matrix multiplications
Use Lapack BLAS if available to compute matrix multiplication and matrix/colVector multiplication.
For instance, a large `(480x640) x (640x480)` matrix multiplication:
 - before: 321 ms
 - after: 9 ms

For more reduced matrix size, 10000 matrix multiplications: `(6x200) x (200x6):`
 - before 96 ms
 - after 46 ms

resolve #190
closed 2017-06-29 by s-trinh