lagadic/visp #206: Add a `.clang-format` file in the top-most directory

xoox, 2017-07-13:
There is already a [ViSP coding style guideline](https://github.com/lagadic/visp/wiki/coding-style-guideline). But codes in the repository is not as consistency as expected.
Some utilities exist to help to enhance coding formats, such as [clang-tidy](http://clang.llvm.org/extra/clang-tidy/) and [clang-format](http://clang.llvm.org/docs/ClangFormat.html).

Would you please add a `.clang-format` file in the top-most directory, which will ensure the coding format to some extents. For example, the following format are used with changes in #205.

```
$ cat .clang-format
```

```yaml
---
Language: Cpp
BasedOnStyle: LLVM
BreakBeforeBraces: Linux
ColumnLimit: 78
ConstructorInitializerIndentWidth: 2
...
```
This configuration has been tested with clang-format-3.9 and clang-format-4.0, and may be compatible with versions >=3.8
```
$ clang-format -i some_source.cpp
```
Wish a `.clang-format` file will be helpful to all developers.

fspindle, 2017-07-13:
This is more an enhancement than an issue.
I didn't know this nice tool. 

I have introduced a `.clang-format` file and also a script that allows to apply the coding rules to all the code. Tested on OSX and Ubuntu 16.04. To use the script: 
````
sudo apt-get install clang-format
cd visp
sh script/format-coding-style.sh`
````

@s-trinh Applying the coding rules to all the code is ok for me. But before doing a commit with all the changes could you check on your side since a lot of things are modified by the tool.

s-trinh, 2017-07-16:
Probably we should tweak a little the `clang-format` options before. Currently, it is mainly based on the [LLVM Coding Standards](https://llvm.org/docs/CodingStandards.html).

Current options:
- `ColumnLimit: 78` too short for me, maybe 100 or 120? 0 will not change anything.
- `SpacesInAngles: true` will put a space for `<>`: `static_cast< int >(arg);` This is not what we do.
- `BreakBeforeBraces: Linux` will always put the brace on the same line: `if () {` but not for the brace after a function definition. Brace on the same line or after is used in the same time in ViSP currently.

Options `IndentWidth` and `UseTab` could be added if useful.

If we adopt the rules in `.clang-format`, the [ViSP coding style guideline](https://github.com/lagadic/visp/wiki/coding-style-guideline) page should be updated accordingly. We should also add the [instructions](https://github.com/andrewseidl/githook-clang-format) to use `clang-format` in an editor and / or to apply the formatting when [committing](https://gist.github.com/wangkuiyi/7379a242f0d4089eaa75).

xoox, 2017-07-17:
*   `ColumnLimit: 78`

It's not recommended to use a large column limit. For `git diff` in
traditional 80 column terminals, 78 is a reasonable maximum choice (80
minus `-/+` sign and a space). Besides, as printing on an A4 paper size
media (PDF or hard-copy), column less than 80 will make sure nicer
format.

You can always use a column of 100 to 120 in your editors with a modern
HD wide screen, just `clang-format` before commit. There's no hurt to
anybody's personal developing experience.

*   `SpacesInAngles: true`

This is used to conform to C++ standard before C++11, where `A<A<int> >`
is used while `A<A<int>>` is preferred after C++11 standard.
An alternative option is `Standard: Cpp03`, or `Standard: Cpp11`
(default in LLVM style) if your compilers won't complain.

*   `BreakBeforeBraces: Linux`

This option overrides the default setting in LLVM style which uses
`BreakBeforeBraces: Attach`. In configuration `BreakBeforeBraces:
Linux`,
like Attach, but break before braces on function, namespace and class
definitions. The only exception is functions with one-line body, which
will be merged to a single line if possible. It's can be controlled by
`AllowShortFunctionsOnASingleLine` option.

The `BasedOnStyle:  LLVM` will set all default options which are similar
to ViSP coding style in most cases. LLVM style can be overridden by
individual options.

In LLVM style, `IndentWidth: 2` and `UseTab: Never` have already been
set. So they are not necessary if `BasedOnStyle:  LLVM` ensures they
conform to ViSP coding style guide.

xoox, 2017-07-21:
If we want no unnecessary space among `<>` instead of `SpacesInAngles:
true`, two choices are there.

*   C++11 style `A<A<int>>`
    -   This is the default effect of `BasedOnStyle: LLVM` which set
        `Standard: Cpp11` internally.
    *   In order to get rid of compiling warnings, CMake option
        `USE_CPP11` should be set default `ON`.

*   C++03 style `A<A<int> >`
    *   Use `Standard: Cpp03`

fspindle, 2017-07-31:
@xoox Thanks

I have replaced `SpacesInAngles: true` with `Standard: Cpp03` that matches better existing ViSP coding style (see commit de9184c3295186472aa51678ca14a9f1ced5edb9)

fspindle, 2018-02-13:
Introduced

closed 2018-02-13 by fspindle