lagadic/visp #207: CMake option 'ACTIVATE_DEBUG_TRACE' has never been used

xoox, 2017-07-19:
There is an option in top level CMakeLists.txt
```
VP_OPTION(ACTIVATE_DEBUG_TRACE  "" "" "Enable debug and trace printings" "" ON)
```

But the variable `ACTIVATE_DEBUG_TRACE` has never been referenced in anywhere. So developers can't enable debug and trace printings globally by CMake.

The following line should be added somewhere, I think.

```
if(ACTIVATE_DEBUG_TRACE)
  add_definitions("-DVP_DEBUG -DVP_TRACE")
endif()
```

fspindle, 2017-07-19:
You are right `ACTIVATE_DEBUG_TRACE` is never used.

The question is, shall we remove or keep this option ?
- From a ViSP user point of view I would say that using `vpTRACE()` is possible following instructions provided in the [tutorial](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-trace.html)
- From a ViSP developper point of view I'm not convinced since modern IDE allow with the debugger to identify a lot of things without instrumenting the code with printings

Bellow my comments if we want to keep this option.

As suggested adding the following lines is not enough since these defs are not propagated when ViSP is used as a 3rd party project (a good practice is to not propagate defs with cmake).
````
if(ACTIVATE_DEBUG_TRACE)
  add_definitions("-DVP_DEBUG -DVP_TRACE")
endif()
````
The solution I see is to set these defs in `vpConfig.h` adding in [CMakeLists.txt](https://github.com/lagadic/visp/blob/master/CMakeLists.txt) after line [267](https://github.com/lagadic/visp/blob/master/CMakeLists.txt#L267)
````
if(ACTIVATE_DEBUG_TRACE)
  set(VP_DEBUG TRUE)
  set(VP_TRACE TRUE)
endif()
````
and modifying [vpConfig.h.in](https://github.com/lagadic/visp/blob/master/cmake/templates/vpConfig.h.in) adding near line [107](https://github.com/lagadic/visp/blob/master/cmake/templates/vpConfig.h.in#L107)
````
// Enable debug and trace printings
#cmakedefine VP_TRACE
#cmakedefine VP_DEBUG
````

This will alllow usage of `vpDEBUG()` and `vpTRACE()` without `VP_DEBUG_MODE`. In other terms, from [tutorial](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-trace.html#example) the following will work:
````
   17   // C-like trace printings if VP_TRACE defined
   18   vpTRACE("C-like trace");              // std::cout
   21   vpERROR_TRACE("C-like error trace");            // std::cerr
   25   vpDEBUG_TRACE ("C-like debug trace"); // stdout
   26   vpDERROR_TRACE("C-like error trace"); // stderr
   32   vpCTRACE << "C++-like trace" << std::endl;       // std::cout
   33   vpCERROR << "C++-like error trace" << std::endl; // std::cerr
   38   vpOUT_FCT("main()"); // std::cout if VP_TRACE defined
````
while the following lines will not be considered:
````
   13   std::cout << "Debug level 1 active: " << vpDEBUG_ENABLE(1) << std::endl;
   14   std::cout << "Debug level 2 active: " << vpDEBUG_ENABLE(2) << std::endl;
   15   std::cout << "Debug level 3 active: " << vpDEBUG_ENABLE(3) << std::endl;
   19   vpTRACE(1, "C-like trace level 1");   // std::cout
   22   vpERROR_TRACE(1, "C-like error trace level 1"); // std::cerr if VP_DEBUG_MODE value is >= 1
   28   vpDEBUG_TRACE (2, "C-like debug trace level 2"); // std::cout if VP_DEBUG_MODE value >= 2
   29   vpDERROR_TRACE(2, "C-like error trace level 2"); // std::cerr if VP_DEBUG_MODE value >= 2
   36   vpCDEBUG(2) << "C++-like debug trace level 2" << std::endl; // std::cout if VP_DEBUG_MODE value >= 2
````
Then maybe we should introduce a new option in the root `CMakeLists.txt` to be able to set `VP_DEBUG_MODE`.

xoox, 2017-07-20:
Thanks for your inspiring thoughts.

*   For users using ViSP as 3rd party library, they should have freedom
    to use or don't use these debug and trace macros. Just define some
    enabling macros if they want.

*   For ViSP developers, there should be a global option to switch
    on/off debug/trace printings. Here's some supporting reasons.
    - Debug/trace macros are used in the source code of ViSP library and
      test applications. e.g., When running some manual testing
      programs, debug and trace information are very helpful indeed.
      + When running `test1394TwoGrabber`, it shows nothing when
        encountering exceptions because of debug macros are disabled.
        There is no option to turn on the debug/trace definitions in
        CMake.
    - Sure modern debuggers are powerful to identify a lot of issues.
      But for some multiple thread applications, time critical services,
      event response or hardware communication apps, log and trace
      information is more helpful in the case of you have no control of
      other peers.

Use either `ACTIVATE_DEBUG_TRACE` or `VP_DEBUG_MODE` is OK. It might be
helpful to add an extra line to your solution.

```diff
  if(ACTIVATE_DEBUG_TRACE)
    set(VP_DEBUG TRUE)
    set(VP_TRACE TRUE)
+   set(VP_DEBUG_MODE TRUE)
  endif()
```

fspindle, 2017-07-20:
Can you have a look to PR #209.

xoox, 2017-07-21:
Nice job.
I tested PR #209 this morning, and found nothing unexpected.
It worked for me to enable debug and trace printings globally.

closed 2017-07-22 by fspindle

fspindle, 2017-07-23:
Fixed with PR #209

PR #209 (2017-07-22): Fix debug/trace behavior
Fixes #207