lagadic/visp #212: No member function setKltMaskBorder() in Class vpMbKltTracker

codetaobeibei, 2017-08-06:
Hello,
 I'm learning the tutorial recently. I ran the example source code tutorial-mb-tracker.cpp in the [tutorial: Markerless model-based tracking ](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-tracking-mb.html), however, the compiler told me that  vpMbKltTracker() is not a member function of the class vpMbKltTracker. And I looked through the header file  vpMbKltTracker.h, it hadn't member function vpMbKltTracker() neither. But I could find this member function on the web tutorial classes [vpMbKltTracker Class Reference](http://visp-doc.inria.fr/doxygen/visp-daily/classvpMbKltTracker.html). So, I don't know what happened with it. Could someone give some hint? By the way, I compiled ViSP within Visual Studio 2015 on Windows 10.

s-trinh, 2017-08-06:
Hello,
Using `setMaskBorder()` instead should solve the issue. `setKltMaskBorder()` has been introduced recently in master but is present in a ViSP release yet.
[Here](http://visp.inria.fr/documentation/) you can find the documentation for each ViSP version.

To know which version of ViSP you are using, you have multiple options:

- open the `ViSP-third-party.txt` file in the ViSP build directory if you have built ViSP from source
- use the following program:

```
#include <iostream>
#include <visp3/core/vpConfig.h>
#include <visp3/core/vpIoTools.h>

#define VALUE_TO_STRING(x) #x
#define VALUE(x) VALUE_TO_STRING(x)

int main() {
  std::cout << vpIoTools::getBuildInformation() << std::endl; //Only since ViSP 3.0.1
  std::cout << "VISP_VERSION: " << VALUE(VISP_VERSION) << std::endl;

  return 0;
}
```

codetaobeibei, 2017-08-06:
Hi s-trinh,
Thanks for your kindly reply. It actually works by using `setMaskBorder()`. And with your hint, I figure out that I compile ViSP of the version 3.0.1, but I learn it with the tutorial version 3.1.0. What a stupid mistake I make! Sorry for wasting your time and thanks again!

closed 2017-08-06 by codetaobeibei