lagadic/visp #240: MBT tutorials：when I want to generate the Executable file‘ tutorial-mb-tracker’，it return errors behind

jch007, 2017-10-09:
/home/jiaochenhang/MBT/src/tutorial-mb-tracker.cpp: In function ‘int main(int, char**)’:
/home/jiaochenhang/MBT/src/tutorial-mb-tracker.cpp:109:47: error: ‘class vpMbKltTracker’ has no member named ‘setKltMaskBorder’
       dynamic_cast<vpMbKltTracker*>(tracker)->setKltMaskBorder(5);
                                               ^
make[2]: *** [src/CMakeFiles/tutorial-mb-tracker.dir/tutorial-mb-tracker.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/tutorial-mb-tracker.dir/all] Error 2
make: *** [all] Error 2

s-trinh, 2017-10-09:
Hello,
`vpMbKltTracker::setKltMaskBorder()` is not present in ViSP 3.0.1 and has been introduced in master recently.
Can you try to use the tutorial code for ViSP 3.0.1: [http://visp.inria.fr/tutorials/](http://visp.inria.fr/tutorials/) instead?

jch007, 2017-10-09:
Thank you for your help

jch007, 2017-10-09:
Thank  you for your help@s-trinh

closed 2017-10-10 by s-trinh