lagadic/visp #249: Tutorial about  Sensor module  -RGB-D sensor ：how to use Kinect in Visp

jch007, 2017-10-19:
I have already installed the  kinect1‘s driver  libfreenect. and I follow the instruction "freenect-glview"to test the kinect1.It can work,but when I test the kinectAcquisition.cpp in tutorial. First ,I generate the exectuable file kinectAcquisition.Second,I want to test with command "sudo ./kinectAcquisition".but it always return the error "You should install libfreenect to run this example ".I want to know how to solve it.look forward to your reply！

jch007, 2017-10-21:
I followed your advice to let`USE_FREENECT ` on and  configure again ViSP with CMake 
If I want to  test the example(do not use official executable file in visp-build example)like

`kinectAcquisition.cpp`

 I write the the  CMakeLists.txt   as follow

`project(kinect)
cmake_minimum_required(VERSION 2.8)
find_package(VISP REQUIRED)
include_directories(${VISP_INCLUDE_DIRS})
add_executable(kinectAcquisition kinectAcquisition.cpp)
target_link_libraries(kinectAcquisition ${VISP_LIBRARIES})`

It can generate the exectuable file successlly.However,it can not work,and error return 
`You should install libfreenect to run this example`

I think something must be wrong with the  CMakeLists.txt .Can you point out where is wrong

fspindle, 2017-12-07:
Do you have seen this page https://visp.inria.fr/3rd_freenect/ ?

fspindle, 2018-01-12:
Some updates ?
Can I close ?

closed 2018-04-01 by fspindle