lagadic/visp #261: Some questions about find contours

YuetianLiu, 2017-11-02:
Recently, I wanted to implement the algorithm of Topological structural analysis of digitized binary images by border following. I found the find contour code in the visp. I used the image format is I(width(),height()), which is just the oppistie of the visp image format. How can I correct the code to fit my image format.

typedef enum {
				NORTH, NORTH_EAST, EAST, SOUTH_EAST, SOUTH, SOUTH_WEST, WEST, NORTH_WEST, LAST_DIRECTION
			} vpDirectionType;

vpDirection() {
		m_dirx[0] = 0; m_dirx[1] = 1;  m_dirx[2] = 1;  m_dirx[3] = 1;
		m_dirx[4] = 0; m_dirx[5] = -1; m_dirx[6] = -1; m_dirx[7] = -1;
	       m_diry[0] = -1; m_diry[1] = -1; m_diry[2] = 0; m_diry[3] = 1;
	       m_diry[4] = 1;  m_diry[5] = 1;  m_diry[6] = 0; m_diry[7] = -1;
	}
				
What do the vpDirectionType ad vpDirection mean? This question has been bothering me for a week.

s-trinh, 2017-11-02:
```
typedef enum {
  NORTH, NORTH_EAST, EAST, SOUTH_EAST, SOUTH, SOUTH_WEST, WEST, NORTH_WEST, LAST_DIRECTION
} vpDirectionType;

vpDirection() {
  m_dirx[0] = 0; m_dirx[1] = 1; m_dirx[2] = 1; m_dirx[3] = 1;
  m_dirx[4] = 0; m_dirx[5] = -1; m_dirx[6] = -1; m_dirx[7] = -1;

  m_diry[0] = -1; m_diry[1] = -1; m_diry[2] = 0; m_diry[3] = 1;
  m_diry[4] = 1; m_diry[5] = 1; m_diry[6] = 0; m_diry[7] = -1;
}
```

These are the pixel offsets for the contour following algorithm:
[image]

ViSP uses this convention:
[image]

If you use another convention, you should be able to convert the pixel coordinates to match your convention.

dldl3D, 2017-11-03:
My convention is as follow:
[image]

dldl3D, 2017-11-03:
My convention is as follows:
[image]

I(1,1)=1, I(1,2)=2, I(2,1)=3, I(2,2)=4

Where should I change the code according my convention? I have been changed some codes, but I failed.

In the vpContours.h, I changed the L120-L121 code 
      int yy = (int) (point.get_j() + m_diry[(int) m_direction]);
      int xx = (int) (point.get_i() + m_dirx[(int) m_direction]);

In pContours.cpp, I changed the  L325-L328 code

  for (unsigned int i = 0; i < I.getWidth(); i++) {
    lnbd = 1; //Reset LNBD at the beginning of each scan row

    for (unsigned int j = 0; j < I.getHeight(); j++) {

Where should I change the code?

s-trinh, 2017-11-03:
For me, the only difference between your convention and ours is that your index start at `1`. We access the 2D array using the row followed by the column index:
```
I[0][0] = 1;
I[0][1] = 2;
I[1][0] = 3;
I[1][1] = 4;
```
I think that you should be able to keep the same code and just add `1` to the pixel coordinates for the `contours` / `contourPts`:
`void vp::findContours (const vpImage< unsigned char > &I_original, vpContour &contours, std::vector< std::vector< vpImagePoint > > &contourPts, const vpContourRetrievalType &retrievalMode = vp::CONTOUR_RETR_TREE)`

dldl3D, 2017-11-03:
My image convention is I (width, height), but yours is I (height, width). should I change the loop?

s-trinh, 2017-11-04:
What data structure do you use as it is strange to start the index from `1` in C++? Looking at the pictures, our image convention is similar to yours except that the index start from `1` in your case.

In ViSP, to access a pixel it is: I[row][col]. In your case, it should be `I(col+1,row+1)` no? If it is the case, the for loops should be changed to (plus some possible other changes needed):
```
for (unsigned int i = 1; i <= I.getHeight(); i++) {
  for (unsigned int j = 1; j <= I.getWidth(); j++) {
  }
}
```

dldl3D, 2017-11-04:
I am so sorry that the pixel also starts 0. I use the CImg( http://cimg.eu ) image data structure. I change the the loop. But I failed.

s-trinh, 2017-11-04:
CImg uses `img(col,row)`, so you just have to take care to supply the same parameter, i.e. when you see `I[5][4]` with `CImg` it should be `img(4,5)`.

See [http://cimg.eu/reference/group__cimg__loops.html#lo4](http://cimg.eu/reference/group__cimg__loops.html#lo4) or [http://cimg.eu/CImg_slides.pdf](http://cimg.eu/CImg_slides.pdf).

dldl3D, 2017-11-04:
Thanks for your help. I have change the parameter. but it doesn’t work. I don’t know where I am wrong. Should I change the vpImagePoint? Change vpImagePoint (xx,yy) to vpImagePoint.get_i() to vpImagePointw.get_j()?

dldl3D, 2017-11-06:
Should I change the code`int pixel = I[yy][xx];` to `int pixel = I(xx,yy)`
`int yy = (int) (point.get_i() + m_diry[(int) m_direction]);` to `int yy = (int) (point.get_i() + m_dirx[(int) m_direction]);`
`from.set_j(from.get_j() + 1);` to `from.set_i(from.get_i() + 1);`

dldl3D, 2017-11-07:
How should I debug was it？

s-trinh, 2017-11-09:
Sorry we cannot help you to debug your code.

> How should I debug was it？

Run the ViSP implementation and yours on a simple example (for [instance](https://github.com/lagadic/visp/blob/master/modules/imgproc/test/testContours.cpp#L264-L289)) and print the states of the variables during the execution. If you use an IDE (Visual Studio, ...), you can use the integrated debugger and breakpoints for this.

More info:
 - `vp::findContours` needs a binary image with only zeros and ones
 - the only change should be to access correctly the CImg matrix / image
 - `row` / `y` / `i` should represent the same thing
 - `col` / `x` / `j` should represent the same thing

Good luck.

closed 2017-11-09 by s-trinh