lagadic/visp #267: Template Matching

OmarHamdoun, 2017-11-16:
Hi
when I use template tracking with (8 DOF )homography , always give me affine transform

fspindle, 2017-12-07:
On my side I'm unable to reproduce this issue.

To check on your side:
- edit `example/tracking/templateTracking.cpp`, 
- modify near line 456
`````
      // Display the template
#if 0
      tracker->display(I, vpColor::red, 3);
#else
      vpTemplateTrackerZone zoneWarped_, zoneRef_ = tracker->getZoneRef();
      vpTemplateTrackerWarp *warp_ = tracker->getWarp();
      vpColVector p_= tracker->getp();
      std::cout << "Model: " << p_.t() << std::endl;
      warp_->warpZone(zoneRef_, p_, zoneWarped_);
      zoneWarped_.display(I, vpColor::red, 3);
      zoneRef_.display(I, vpColor::green, 3);
#endif
````
- build 
- run `./templateTracker -w 1` 

The displayed model should have 8 parameters (homography), while with `-w 0` only 6 (affine)

fspindle, 2017-12-07:
The tutorial `tutorial/tracking/template-tracker\tutorial-template-tracker.cpp` seems also working

OmarHamdoun, 2017-12-07:
thanks
but my problem is the third line of matrix is always zeros, so we haven't any rotions round x axis or y axis

fspindle, 2017-12-07:
If the example and tutorial are working, it means that you have maybe a bug in your code !

Which tracker class are you using ?

OmarHamdoun, 2017-12-07:
hi
I have just modified templateTracking.cpp  for calculate   homography  matrix from P, i have just added these lines:
      // Display the template
#if 0
      tracker->display(I, vpColor::red, 3);
#else
      vpTemplateTrackerZone zoneWarped_, zoneRef_ = tracker->getZoneRef();
      vpTemplateTrackerWarp *warp_ = tracker->getWarp();
      vpColVector p_= tracker->getp();
	  std::cout << "p="<< p_ << std::endl;
	  
      warp_->warpZone(zoneRef_, p_, zoneWarped_);
      zoneWarped_.display(I, vpColor::red, 3);
      zoneRef_.display(I, vpColor::green, 3);
	  vpHomography aHb;
	  aHb[0][0] = 1 + p_[0]; aHb[0][1] = p_[3]; aHb[0][2] = p_[6];
	  aHb[1][0] = p_[1]; aHb[1][1] = 1+p_[4]; aHb[1][2] = p_[7];
	  aHb[2][0] = p_[2]; aHb[2][1] = p_[5]; aHb[2][2] = 1;
	  std::cout << "p=" << p_ << std::endl;
	  vpRotationMatrix aRb;	vpTranslationVector atb;	vpColVector n;
	  vpThetaUVector atub;
	  aHb.computeDisplacement(aRb, atb, n);
	  atub.buildFrom(aRb);
	  vpColVector Displacement(6);
	  for (unsigned int i = 0; i < 3; i++)
	  {
		  Displacement[i] = atb[i];
		  Displacement[i + 3] = atub[i];
	  }
	  if (true){
		  std::cout << "\nEstimated displacement Homography:" << std::endl;
		  std::cout << "Translation of Homography: ";
		  for (int i = 0; i < 3; i++)
			  std::cout << Displacement[i] << " ,  "; std::cout << std::endl;
		  std::cout << "Rotation of Homography deg :";
		  for (int i = 0; i < 3; i++)
			  std::cout << vpMath::deg(Displacement[i + 3]) << ",  "; std::cout << std::endl;
	  }
#endif
I used the following  tracker and warper:
	TrackerType opt_tracker_type = TRACKER_ZNCC_INVERSE_COMPOSITIONAL;
	WarpType opt_warp_type = WARP_HOMOGRAPHY;
thanks for your interesting

fspindle, 2017-12-07:
If I introduce the modifications you mention and launch `$ ./templateTracker -t 5 -w 1`
I get the following results showing that the estimaped model (homography) has 8 parameters as expected.

`````
Process image number 30
p=-0.05513767992
-0.02469646194
2.106050274e-05
0.01659110889
-0.06718808238
-6.282449086e-05
14.22721214
-11.46253754
p=-0.05513767992
-0.02469646194
2.106050274e-05
0.01659110889
-0.06718808238
-6.282449086e-05
14.22721214
-11.46253754

Estimated displacement Homography:
Translation of Homography: 15.23700219 ,  -12.27620181 ,  0.07097809839 ,  
Rotation of Homography deg :-0.004116772416,  0.001962802812,  -0.9584537997,
`````

These results are good.

OmarHamdoun, 2017-12-07:
hi
 the value of rotation around x and y is very small in spite of the rotation of cube is clear
the  rotation around Z gives reasonable result but not the same around X & Y. the rotation around X give translation more than rotation
thanks

closed 2017-12-11 by fspindle