lagadic/visp #271: Linking CXX problem

changkaizhao, 2017-12-11:
I have Linking problem as below:
```
[ 17%] Building CXX object modules/core/CMakeFiles/visp_core.dir/src/tracking/vpTracker.cpp.o
Linking CXX shared library ../../lib/libvisp_core.dylib
ld: warning: could not create compact unwind for _dormqr_: stack subq instruction is too different from dwarf stack size
ld: warning: could not create compact unwind for _dormlq_: stack subq instruction is too different from dwarf stack size
Undefined symbols for architecture x86_64:
  "__gfortran_concat_string", referenced from:
      _dormqr_ in liblapack.a(dormqr.o)
      _dtrtri_ in liblapack.a(dtrtri.o)
      _dormbr_ in liblapack.a(dormbr.o)
      _dormlq_ in liblapack.a(dormlq.o)
  "__gfortran_st_write", referenced from:
      _xerbla_ in liblapack.a(xerbla.o)
  "__gfortran_st_write_done", referenced from:
      _xerbla_ in liblapack.a(xerbla.o)
  "__gfortran_stop_string", referenced from:
      _xerbla_ in liblapack.a(xerbla.o)
  "__gfortran_string_len_trim", referenced from:
      _xerbla_ in liblapack.a(xerbla.o)
  "__gfortran_transfer_character_write", referenced from:
      _xerbla_ in liblapack.a(xerbla.o)
  "__gfortran_transfer_integer_write", referenced from:
      _xerbla_ in liblapack.a(xerbla.o)
ld: symbol(s) not found for architecture x86_64

```

Cmake compiler settings as below:
```
  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                /usr/bin/c++  (ver 9.0.0.9000038)
    C++ flags (Release):         -Wall -Wextra -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C++ flags (Debug):           -Wall -Wextra -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -Wall -Wextra -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -O3 -DNDEBUG
    C flags (Debug):             -Wall -Wextra -fvisibility=hidden -msse2 -msse3 -mssse3 -fPIC -g
    Linker flags (Release):      
    Linker flags (Debug):        

```
 OS is `Mac OS 10.12.6`  
`clang -v`
```
Apple LLVM version 9.0.0 (clang-900.0.38)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
Found CUDA installation: /usr/local/cuda, version 8.0
```

fspindle, 2017-12-11:
Which ViSP version are you using ?
What is the result of `gfortran -v`

Run `cmake ../visp` and put the complete content of `ViSP-third-party.txt` file in your answer.

changkaizhao, 2017-12-12:
`visp` is fetched from the master branch on Github.

`gfortran -v`
```
gfortran: warning: couldn’t understand kern.os version ‘16.7.0
Using built-in specs.
COLLECT_GCC=gfortran
COLLECT_LTO_WRAPPER=/usr/local/gfortran/libexec/gcc/x86_64-apple-darwin13/4.9.0/lto-wrapper
Target: x86_64-apple-darwin13
Configured with: ../gcc-4.9.0/configure --prefix=/usr/local/gfortran --with-gmp=/Users/fx/devel/gcc/deps-static/x86_64 --build=x86_64-apple-darwin13 --enable-languages=c,c++,fortran,objc,obj-c++
Thread model: posix
gcc version 4.9.0 (GCC) 
```

fspindle, 2017-12-12:
What is the output of `gcc --version`?

changkaizhao, 2017-12-12:
```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.0.0 (clang-900.0.38)
Target: x86_64-apple-darwin16.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

changkaizhao, 2017-12-12:
I use `brew install` to install it successfully. thanks.

fspindle, 2017-12-12:
I suspect that the version of gfortran based on gcc 4.9.0 is uncompatible with gcc 4.2.1.
In other terms gcc and gfortran should be based on the same version.

Maybe a `brew update` and then `brew upgrade` should make things working.

closed 2018-01-12 by fspindle