lagadic/visp #272: How to transform the velocity computed in IBVS from \thetaU to Quaternion

TokyoClod, 2017-12-11:
Form the function 
`vpColVector vpServo::computeControlLaw()` 
I get the velocity in camera frame in the form of thetaU  
I am wondering how to transform the velocity from  thetaU form to Quaternion form so I can use it in ros.

fspindle, 2017-12-11:
The `vpColVector` returned by `vpServo::computeControlLaw()` is a velocity twist (3 translations velocities vx,vy,y, + 3 rotations velocities wx,wy,wz) corresponding to the camera velocity.

In ROS you can convert this data structure:

````
geometry_msgs::TwistStampedConstPtr msg;
servo.setServo(vpServo::EYEINHAND_CAMERA);
...
vpColVector vc = servo.computeControlLaw();
msg->twist.linear.x = vc[0];
msg->twist.linear.y = vc[1];
msg->twist.linear.z = vc[2];

msg->twist.angular.x = vc[3];
msg->twist.angular.y = vc[4];
msg->twist.angular.z = vc[5];
````

TokyoClod, 2017-12-11:
From the description in  [http://visp-doc.inria.fr/doxygen/visp-daily/classvpSimulatorCamera.html#a19368e64873f93c807a1bfe0c6f102df](http://visp-doc.inria.fr/doxygen/visp-daily/classvpSimulatorCamera.html#a19368e64873f93c807a1bfe0c6f102df)
I think the form of rotations velocities is [vpThetaUVector](http://visp-doc.inria.fr/doxygen/visp-daily/classvpThetaUVector.html), 
so I am wondering if it is right just pass the  vc[3]...vc[5] to geometry_msgs::Twist

fspindle, 2017-12-11:
The doc  http://visp-doc.inria.fr/doxygen/visp-daily/classvpSimulatorCamera.html#a19368e64873f93c807a1bfe0c6f102df is completely wrong. I will fix it.

`vpThetaUVector`represents an angular position, nor a velocity.
You may do as described in my previous post.

commit e5cdcb12 (2017-12-11): Fix setVelocity() documentation. Closes #272.
closed 2017-12-11 by fspindle