lagadic/visp #276: vpDisplay::getKeyBoardEvent() can not always get key board event ?

codetaobeibei, 2017-12-21:
Hi,
   I use `vpDisplay::getKeyBoardEvent()` in a loop for camera frame capturing, and I set some keys to do other thing ,like stop the loop or save a frame from camera. However, sometimes when I press the keys, `vpDisplay::getKeyBoardEvent() ` does not always work -which means that sometimes I press several times, the function works one time. I do not sure whether it is my code problem or something else.
  Below is my code wrt the key board event:
```
int main(int argc, char** argv)
{
......do other things
        vpRGBa Ir;
	char key[3] = { 'q','c','d' };
......do other things
       while(cam.isOpend())
       {
                cap >> frame;
		vpImageConvert::convert(frame, Ir);
                .....
		vpDisplay::display(Ir);
		vpDisplay::flush(Ir);
		bool event = vpDisplay::getKeyboardEvent(Ir, &key[0], false);
		if(event)
			std::cout << "key: " << key << std::endl;
		vpTime::wait(5);
       }
......do other things
}
```

s-trinh, 2017-12-21:
The issue appears when the image is focused?

codetaobeibei, 2017-12-21:
Hi, @s-trinh  . The camera I use can not autofocus and has been calibrated.

s-trinh, 2017-12-21:
Sorry, I meant: do you have the issue when the display window is focused (display window is selected and on top) as it could just be because the display window that should register the key event is not selected, has not the focus.

codetaobeibei, 2017-12-21:
Sorry for my misunderstanding. I am sure that I select the display window and the key event has been registered. Because when I press the key for a long time(not press and then release suddenly), the console will print Key words serveral times 'slowly'. By the way,I just test that it cost 100ms(average) for one loop because of the tag detection(not sure if this will effect) .

s-trinh, 2017-12-21:
Yes probably it is due to the time spent to detect the AprilTag. Just comment the detection to see if the issue disappear or not.

You can try to change the [`setAprilTagQuadDecimate()`](http://visp-doc.inria.fr/doxygen/visp-daily/classvpDetectorAprilTag.html#a9fcd4eec9276f16a4d3a7689fb27696c) factor to reduce the computation time but the quality of the detection will decrease.

`vpTime::wait(5);` looks not necessary in your code as the capture should be blocking.

Probably the solution should be to use two distinct threads: one for the display and the key event registration (a non blocking thread) and another one for the AprilTag detection. But you will need to implement some logic to check if the detection thread has finished before submitting a new image and get the result of the detection.

codetaobeibei, 2017-12-21:
Thanks for your advise. However, it does not work still. I set `setAprilTagQuadDecimate(2)` and  `setAprilTagNbThreads(1)` before the loop of detection. And if I remove `vpTime::wait(5);`, the issue still happens.
I'm not sure whether it depends on defferent PC or personal configuration. when I use another laptop, it always print "key: 255"(for every frame). I use the code below to test(OS:Windows 10, ViSP version: 3.0.1):
```
int main(int argc, const char** argv)
{
	int opt_device = 0;
	vpImage<unsigned char> I;
	cv::VideoCapture cap(opt_device);
	if (!cap.isOpened()) { // check if we succeeded
		std::cout << "Failed to open the camera" << std::endl;
		return EXIT_FAILURE;
	}
	char key[3] = { 'q','c','d' };
	cv::Mat frame;
	cap >> frame; // get a new frame from camera
	vpImageConvert::convert(frame, I);
	vpDisplayOpenCV d(I);
	for(;;)
	{
		cap >> frame;
		vpImageConvert::convert(frame, I);
		vpDisplay::display(I);
		Sleep(50);//some detect cost
		vpDisplay::flush(I);
		if(vpDisplay::getKeyboardEvent(I, &key[0], false))
		{
			std::cout << "Key: " << key << std::endl;
		}	
	}
	return 0;
}
```
I will try to devide the detect and display and key event registration into two defferent threads tomorrow.

codetaobeibei, 2017-12-21:
I am not sure whether I use the get key board function in a wrong way. if I just change `vpDisplay::getKeyboardEvent(I, &key[0], false)` to `vpDisplay::getKeyboardEvent(I, &key[1], false)` . The rest code is still the same above. The print change form "Key: 255" to "Key: q255" for every frame without any key board pressed.

s-trinh, 2017-12-21:
Indeed, I have the same issue with the mentioned code. I don't have the issue with `vpDisplayX` (only on Unix) instead of `vpDisplayOpenCV`. Maybe you can try with `vpDisplayGDI` for Windows?

Alternative could be to use instead [`vpKeyboard`](http://visp-doc.inria.fr/doxygen/visp-daily/classvpKeyboard.html).

codetaobeibei, 2017-12-22:
I use `vpDisplayGDI` and it works! Thank you!Although whatever key I press-even out of the `char key[3] = { 'q','c','d' };` I defined-it will print it out. At least, I can use it. Maybe there has some other implemantation problems with mine.

s-trinh, 2017-12-22:
Good news!

That is because in `vpDisplay::getKeyboardEvent(I, &key[0], false)`, `key` is just a char buffer to hold the result of the key pressed, not a "key selector". See the [doc](http://visp-doc.inria.fr/doxygen/visp-daily/classvpDisplay.html#a954a7cffe72e75403d31389a2d1ee5a9).

You can use also:
```
std::string key = "";
if (vpDisplay::getKeyboardEvent(I, key, false)) {
  std::cout << "Key detected: " << key << std::endl;
}
```

codetaobeibei, 2017-12-22:
I get it and there's no other problem of using key board event. Thanks again for your patient reply and explanation!

closed 2017-12-22 by codetaobeibei