lagadic/visp #292: How to convert a 3d model from solidworks to BND file?

codetaobeibei, 2018-01-26:
Hi,
I have a stp file from Solidworks, I want to conert it to BND file which is used in your vpScene.cpp.But I failed to convert it because of no normal tool to do this work. So, could you please give some tip or share some methods about it?Thank you.

fspindle, 2018-01-26:
I just make available [sld2bnd](https://github.com/lagadic/sld2bnd) project that allows to convert Solidworks `*.sld` files to `*.bnd` files compatible with the robot simulator in ViSP

codetaobeibei, 2018-01-27:
Thank you very much for your kindly reply and sharing the project. It is very useful for me. I am trying the method you give, but a problem comes to me is that my Solidworks(version 2016) can not export `*.stp` files as `*.sld` files. Or do you mean `*.sldprt`? Could you please tell me which version of solidworks you are using? Thank you.

fspindle, 2018-01-29:
I don't know. You should check if the content of your `*.sldprt` is similar to the sample files `*.sld` that are in ViSP robot module. 

The modeling of the Afma6 and Viper850 robot simulators was done by hand without Solidworks

closed 2018-04-09 by codetaobeibei