lagadic/visp #324: How to compile ViSP within Visual Studio 2013 on windows 7?

codetaobeibei, 2018-04-09:
Hi,
   I am using ViSP for my projects. But for some reasons, I have to transfer my project from windows 10 to windows 7. Since the installation tutorials in document are all about intalling it on windows 10, and I tried to compile it on windows 7 but failed. So, could you please give some tips about how to compile ViSP 3.1 on windows 7? Thang you.

fspindle, 2018-04-09:
There is this [tutorial](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-install-win7.html) for win7.

If it doesn't help you have to indicate:
- visp release version or from git 
- what is going wrong with a picture or logs...

codetaobeibei, 2018-04-09:
I tried serveral "snapshot" versions, and they all failed, it is seems that the errors come from visp_core project and visp_apriltag project. 
The errors are mainly just like following:
`error LNK2001:  Unresolved external symbols "public: __cdecl vpColVector::vpColVector(class vpColVector const &,unsigned int,unsigned int)" (??0vpColVector@@QEAA@AEBV0@II@Z)	G:\ViSP\build\modules\core\vpPixelMeterConversion.obj	visp_core
error LNK2001: Unresolved external symbols "public: class vpRowVector & __cdecl vpRowVector::operator*=(double)" (??XvpRowVector@@QEAAAEAV0@N@Z)	G:\ViSP\build\modules\core\vpMeterPixelConversion.obj	visp_core
error C2275: “zarray_t”:  illegal use of this type as an expression G:\ViSP\sources\3rdparty\apriltag\apriltag.c	1175  1 visp_apriltag
error C2065: “quads”: undeclared identifier G:\ViSP\sources\3rdparty\apriltag\apriltag.c	1229	1	visp_apriltag`

codetaobeibei, 2018-04-09:
I just tried again the version 3.1, it comes to the following errors:

error C2065: “detections”: undeclared identifier	G:\ViSP\sources\3rdparty\apriltag\apriltag.c	1386	1	visp_apriltag

error C2065: “quads”: undeclared identifier	G:\ViSP\sources\3rdparty\apriltag\apriltag.c	1175	1	visp_apriltag

error C2275: “zarray_t”: illegal use of this type as an expression G:\ViSP\sources\3rdparty\apriltag\apriltag.c	1194	1	visp_apriltag

error LNK1104: cannot open file“..\..\..\..\x64\vc12\lib\Debug\visp_detection310d.lib”	G:\ViSP\build\tutorial\tracking\model-based\generic\LINK	tutorial-mb-tracker

fspindle, 2018-04-09:
Strange. The error comes from `zarray_t` defined in `zarray.h` that is already included in `apriltag.c`.

Could you try to modify `zarray.h` moving line 52 after `struct zarray` as following:

````
struct zarray
{
    size_t el_sz; // size of each element

    int size; // how many elements?
    int alloc; // we've allocated storage for how many elements?
    char *data;
};
typedef struct zarray zarray_t;
````

If it doesn't build, I don't have any other idea for the moment than turning `WITH_APRILTAG=OFF` in CMake GUI, configure and try a new build. Let me know.

codetaobeibei, 2018-04-09:
I tried to modify the lines you metioned, but it still doesn't work, the error are still the same above.  With turning `WITH_APRILTAG=OFF`, it can compile successifully, however, not only the core of ViSP, but also the 3rd party of Apriltag is used in my project.

fspindle, 2018-04-16:
Which are the 3rd parties detected during configuration in CMakeGUI ?
Put the content of `ViSP-third-party.txt` file.

codetaobeibei, 2018-04-16:
Thank you very much for your reply!
Below is the content of the file.
```
==========================================================
General configuration information for ViSP 3.1.0

  Version control:               unknown

  Platform:
    Timestamp:                   2018-04-16T06:55:46Z
    Host:                        Windows 6.1.7601 AMD64
    CMake:                       3.10.3
    CMake generator:             Visual Studio 12 2013 Win64
    CMake build tool:            C:/Program Files (x86)/MSBuild/12.0/bin/MSBuild.exe
    MSVC:                        1800

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                D:/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe  (ver 18.0.21005.1)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MP8   -openmp /MD /O2 /Ob2 /DNDEBUG
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MP8   -openmp /MDd /Zi /Ob0 /Od /RTC1
    C Compiler:                  D:/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3   /MP8   -openmp /MD /O2 /Ob2 /DNDEBUG
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3   /MP8   -openmp /MDd /Zi /Ob0 /Od /RTC1
    Linker flags (Release):      /machine:x64 /INCREMENTAL:NO
    Linker flags (Debug):        /machine:x64 /debug /INCREMENTAL

  ViSP modules:
    To be built:                 core gui imgproc io klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 -

  Windows RT support:            no

  Build options: 
    Build deprecated:            yes
    Build with moment combine:   no

  Mathematics: 
    Use Lapack/blas:             no
    Use Lapack (built-in):       yes (ver 3.2.1)
    Use Eigen3:                  no
    Use OpenCV:                  yes (ver 3.2.0)
    Use GSL:                     no

  Simulator: 
    Ogre simulator: 
    \- Use Ogre3D:               no
    \- Use OIS:                  no
    Coin simulator: 
    \- Use Coin3D:               no
    \- Use SoWin:                no
    \- Use SoXt:                 no
    \- Use SoQt:                 no
    \- Use Qt4:                  no
    \- Use Qt3:                  no

  Media I/O: 
    Use JPEG:                    no
    Use PNG:                     no
    \- Use ZLIB:                 no
    Use OpenCV:                  yes (ver 3.2.0)

  Real robots: 
    Use Afma4:                   no
    Use Afma6:                   no
    Use Franka:                  no
    Use Viper650:                no
    Use Viper850:                no
    Use aria (Pioneer):          no
    Use PTU46:                   no
    Use Biclops PT:              no

  GUI: 
    Use X11:                     no
    Use GTK:                     no
    Use OpenCV:                  yes (ver 3.2.0)
    Use GDI:                     yes
    Use Direct3D:                no

  Cameras: 
    Use DC1394-2.x:              no
    Use CMU 1394:                no
    Use V4L2:                    no
    Use directshow:              no
    Use OpenCV:                  yes (ver 3.2.0)
    Use Flycapture:              no
    Use Pylon:                   no

  RGB-D sensors: 
    Use Realsense:               no
    Use Realsense2:              no
    \- Use PCL:                  no
       \- Use VTK:               no
    Use Kinect:                  no
    \- Use libfreenect:          no
    \- Use libusb-1:             no
    \- Use pthread:              no

  F/T sensors: 
    Use atidaq (built-in):       no
    Use comedi:                  no

  Detection: 
    Use zbar:                    no
    Use dmtx:                    no
    Use AprilTag (built-in):     yes (ver 0.9.8)

  Misc: 
    Use Clipper (built-in):      yes (ver 6.4.2)
    Use XML2:                    yes (ver 2.9.7)

  Optimization: 
    Use OpenMP:                  yes
    Use pthread:                 no
    Use pthread (built-in):      yes (ver 2.9.1)
    Use cxx11:                   no

  Documentation: 
    Use doxygen:                 no

  Tests and samples:
    Tests:                       yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes

  Install path:                  G:/ViSP/build/install

==========================================================
```

fspindle, 2018-04-16:
Just to be sure that the build issue doesn't come from external 3rd party, could you turn OpenCV and XML2 OFF in CMakeGUI, configure and try a new build. Let me know

codetaobeibei, 2018-04-16:
It seems not the problem from external 3rd party.I use the version 3.1.0 to bulid it.
I just turn two parties you mentioned, below is the `ViSP-third-party.txt` file.

```
==========================================================
General configuration information for ViSP 3.1.0

  Version control:               unknown

  Platform:
    Timestamp:                   2018-04-16T07:13:00Z
    Host:                        Windows 6.1.7601 AMD64
    CMake:                       3.10.3
    CMake generator:             Visual Studio 12 2013 Win64
    CMake build tool:            C:/Program Files (x86)/MSBuild/12.0/bin/MSBuild.exe
    MSVC:                        1800

  C/C++:
    Built as dynamic libs?:      yes
    C++ Compiler:                D:/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe  (ver 18.0.21005.1)
    C++ flags (Release):         /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MP8   -openmp /MD /O2 /Ob2 /DNDEBUG
    C++ flags (Debug):           /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MP8   -openmp /MDd /Zi /Ob0 /Od /RTC1
    C Compiler:                  D:/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe
    C flags (Release):           /DWIN32 /D_WINDOWS /W3   /MP8   -openmp /MD /O2 /Ob2 /DNDEBUG
    C flags (Debug):             /DWIN32 /D_WINDOWS /W3   /MP8   -openmp /MDd /Zi /Ob0 /Od /RTC1
    Linker flags (Release):      /machine:x64 /INCREMENTAL:NO
    Linker flags (Debug):        /machine:x64 /debug /INCREMENTAL

  ViSP modules:
    To be built:                 core gui imgproc io klt me sensor ar blob robot visual_features vs vision detection mbt tt tt_mi
    Disabled:                    -
    Disabled by dependency:      -
    Unavailable:                 -

  Windows RT support:            no

  Build options: 
    Build deprecated:            yes
    Build with moment combine:   no

  Mathematics: 
    Use Lapack/blas:             no
    Use Lapack (built-in):       yes (ver 3.2.1)
    Use Eigen3:                  no
    Use OpenCV:                  no
    Use GSL:                     no

  Simulator: 
    Ogre simulator: 
    \- Use Ogre3D:               no
    \- Use OIS:                  no
    Coin simulator: 
    \- Use Coin3D:               no
    \- Use SoWin:                no
    \- Use SoXt:                 no
    \- Use SoQt:                 no
    \- Use Qt4:                  no
    \- Use Qt3:                  no

  Media I/O: 
    Use JPEG:                    no
    Use PNG:                     no
    \- Use ZLIB:                 no
    Use OpenCV:                  no

  Real robots: 
    Use Afma4:                   no
    Use Afma6:                   no
    Use Franka:                  no
    Use Viper650:                no
    Use Viper850:                no
    Use aria (Pioneer):          no
    Use PTU46:                   no
    Use Biclops PT:              no

  GUI: 
    Use X11:                     no
    Use GTK:                     no
    Use OpenCV:                  no
    Use GDI:                     yes
    Use Direct3D:                no

  Cameras: 
    Use DC1394-2.x:              no
    Use CMU 1394:                no
    Use V4L2:                    no
    Use directshow:              no
    Use OpenCV:                  no
    Use Flycapture:              no
    Use Pylon:                   no

  RGB-D sensors: 
    Use Realsense:               no
    Use Realsense2:              no
    \- Use PCL:                  no
       \- Use VTK:               no
    Use Kinect:                  no
    \- Use libfreenect:          no
    \- Use libusb-1:             no
    \- Use pthread:              no

  F/T sensors: 
    Use atidaq (built-in):       no
    Use comedi:                  no

  Detection: 
    Use zbar:                    no
    Use dmtx:                    no
    Use AprilTag (built-in):     yes (ver 0.9.8)

  Misc: 
    Use Clipper (built-in):      no
    Use XML2:                    no

  Optimization: 
    Use OpenMP:                  yes
    Use pthread:                 no
    Use pthread (built-in):      yes (ver 2.9.1)
    Use cxx11:                   no

  Documentation: 
    Use doxygen:                 no

  Tests and samples:
    Tests:                       yes
    Demos:                       yes
    Examples:                    yes
    Tutorials:                   yes

  Install path:                  G:/ViSP/build/install

==========================================================
```
and the error under Debug Mode is still the following.

```
error C2065: “detections”: undeclared identifier	G:\ViSP\sources\3rdparty\apriltag\apriltag.c	1386	1	visp_apriltag

error C2065: “quads”: undeclared identifier	G:\ViSP\sources\3rdparty\apriltag\apriltag.c	1175	1	visp_apriltag

error C2275: “zarray_t”: illegal use of this type as an expression 
 G:\ViSP\sources\3rdparty\apriltag\apriltag.c	   1194	1	visp_apriltag

error LNK1104: cannot open file“..\..\..\..\x64\vc12\lib\Debug\visp_detection310d.lib”	G:\ViSP\build\tutorial\tracking\model-based\generic\LINK	tutorial-mb-tracker
```

fspindle, 2018-04-16:
@codetaobeibei  Could you try in `Release`

@s-trinh Do you have an idea ?

fspindle, 2018-04-16:
I don't have a computer under win7.

fspindle, 2018-04-19:
I succeed to reproduce the issue on Visual 2012 (msvc11). I will check.

closed 2018-04-20 by fspindle

codetaobeibei, 2018-04-20:
I just recently see the good news and really grateful to you!

fspindle, 2018-04-20:
Thanks. It was not easy !

PR #335 (2018-04-20): Fix apriltag msvc
Fix consists in building apriltag library with a CXX compiler.
Closes #324