lagadic/visp #325: Can I ask you about the camera model?

yongjiaPython, 2018-04-10:
## Hello @fspindle, I want to build a virtual servoing platform. Camera model confused me.  So，Can you tell me the camera model used for the sample program experiment? Thank you for your help！

fspindle, 2018-04-10:
The model is explained in this [tutorial](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-calibration-intrinsic.html)

In simulation, a good starting is to set u_0 = (image_width)  / 2 and v_0 = (image height) / 2. You can consider px = py = 600.

closed 2018-04-10 by fspindle

yongjiaPython, 2018-04-11:
## Hello @fspindle，I want to know the camera brand, parameters, and type not the camera model. I want to buy a camera like your experiment used. Thank you for your help！