lagadic/visp #342: AprilTag on iOS. live camera and the distance.

aYasutomo, 2018-04-26:
Hi all,

thank you for your great work.
I'm very interested in AprilTag on iOS.

I would like to ask that is it possible to process live camera image with your AprilTag framework?

Thank you for your guide, I've done it on a prepared image.
http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-apriltag-ios.html
But I'm wondering if I can do it on live camera.

In addition to that, I also would like to ask whether I can know the distance between the tag and camera. In theory, I think I could. I want to know I can do it with your framework.

Sincerely.

aYasutomo, 2018-05-01:
Dear @fspindle,

Thank you for your kind reply, I've been able to detect AprilTag on iOS live camera! :+1:

but It seems my application can't estimate the pose.
Would it be possible to give me some advice?

### Current problem
I can detect AprilTag using this:
~~~~
detector.detect(I);
~~~~
but when I use this line below instead, my application suddenly stops at the same time as the camera captures AprilTag:
~~~~
detector.detect(I, tagSize, cam, cMo_vec);
~~~~

This is the error sentence:
~~~~
libc++abi.dylib: terminating with uncaught exception of type vpMatrixException:
The algorithm computing SVD failed to converge.
~~~~

### I think:
It seems my application is using **VISP_HAVE_LAPACK** instead of **VISP_HAVE_OPENCV_VERSION** in vpMtrix.cpp, since the error sentence comes from  `vpMatrix::svdLapack` in vpMatrix_svd.cpp. How can I use `vpMatrix::svdOpenCV`?

### this is the source code:
~~~ cpp
// Convert image to visp
vpImage<unsigned char> I;
vpImageConvert::convert(mat, I);

// Detect AprilTag
vpDetectorAprilTag::vpAprilTagFamily tagFamily = vpDetectorAprilTag::TAG_36h11;
vpDetectorAprilTag::vpPoseEstimationMethod poseEstimationMethod = vpDetectorAprilTag::HOMOGRAPHY_VIRTUAL_VS;
double tagSize = 0.003;
float quad_decimate = 1.0;
int nThreads = 1;
std::vector<vpHomogeneousMatrix> cMo_vec;

// Set camera parameters
vpCameraParameters cam;
cam.initPersProjWithoutDistortion(615.1674805, 615.1675415, 312.1889954, 243.4373779);

// Initialize apriltag detector
vpDetectorAprilTag detector(tagFamily);
detector.setAprilTagQuadDecimate(quad_decimate);
detector.setAprilTagPoseEstimationMethod(poseEstimationMethod);
detector.setAprilTagNbThreads(nThreads);    

// Detect all the tags in the image
detector.detect(I, tagSize, cam, cMo_vec);       // <-- stops here.
// detector.detect(I);

// Parse detected tags for printings
std::cout << "Number of tags in the image: " << detector.getNbObjects() << std::endl;
for(size_t i=0; i < detector.getNbObjects(); i++) {
    std::cout << "- Detected tag: " << detector.getMessage(i) << std::endl;
    // std::cout << "  pose: " << cMo_vec[i] << std::endl;
}
~~~

Best regards.

s-trinh, 2018-05-02:
You probably need to update these values to your camera.

> cam.initPersProjWithoutDistortion(615.1674805, 615.1675415, 312.1889954, 243.4373779);

 Have a look at this [tutorial](http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-calibration-intrinsic.html).

aYasutomo, 2018-05-06:
I'll see it. thank you!

fspindle, 2018-05-06:
If it works, feel free to contribute proposing for example a tutorial ;-)

aYasutomo, 2018-05-16:
Hi @fspindle and @s-trinh,
thank you for you guides, I can detect AprilTags with live camera!!

I've got cMo_vec values like below. How can I compute the distance between tags and camera?
I can't find the information about cMo_vec.

```
0.949077179  -0.2670242266  0.1671842416  -0.2215270889
-0.2056482543  -0.9271088368  -0.3133336883  -0.1780527684
0.2386656736  0.2629967056  -0.9348109055  0.2070037374
```

[image]

fspindle, 2018-05-16:
As described in the `vpHomogeneousMatrix` class [doc](http://visp-doc.inria.fr/doxygen/visp-daily/classvpHomogeneousMatrix.html), the last column of the matrix is the translation vector tx, ty, tz. The distance is then given by sqrt(tx*tx + ty*ty + tz*tz)

aYasutomo, 2018-05-16:
Thank you as always.

closed 2018-05-16 by aYasutomo

jcnwholland, 2021-06-07:
Hi, i'm totally new to visp, but ran into it trying to improve camera tracking, after using apriltags in a simple way for guidance. Having not many program skills ( I am a medical doctor during daytime), I was wondering if it is possible to realtime safe the position data with the camera as origin, while saving the input footage at the same time? I've got a dual boot system here, what do you recomment? The april tag IOS app is running, but its the downstream process.

thxn in advance, Joep
JCN Willemsen, MD PhD

SachinSAgrawal, 2023-07-20:
I know that this is a bit old, but where is the full code for this? I want to implement something like this into one of my apps.

jcnwholland, 2023-07-22:
what part? the java receive bit over udp?

SachinSAgrawal, 2023-07-22:
Tracking AprilTags with iOS. I tried following this guide https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-apriltag-ios.html, but I am stuck on the first step as the Xcode project link is invalid.

s-trinh, 2023-07-22:
@SachinSAgrawal  The project should be here: https://github.com/lagadic/visp/tree/master/tutorial/ios/StartedAprilTag

jcnwholland, 2023-07-22:
Hmm the official apriltag rep removed the osx rapper, dont have experience with xcode, but maybe this could help your quest: [](https://github.com/AplixCorporation/groma-AprilTag)

SachinSAgrawal, 2023-07-23:
Thanks! Someone commented yesterday but then deleted their message (?) saying that the link was actually https://github.com/lagadic/visp/tree/master/tutorial/ios/StartedAprilTag. I will check out both.