lagadic/visp #367: ViSP installation on Ubuntu 18.04

AKS1996, 2018-05-28:
Has anyone tried a fresh install on Ubuntu 18.04? I'm sure `opencv` is missing a lot of libraries, whose support had been removed from 16.10 onwards.

fspindle, 2018-05-28:
I tried a couple of weeks ago and it was working using `libopencv-dev` package.
Do you have some issues ?

AKS1996, 2018-05-28:
Recently I switched from 16.04 to 18.04. Earlier the tutorial's were compiling well. Now, while compiling the the webcam video capture `vpAprilTag` tutorial, it shows
```
/usr/bin/x86_64-linux-gnu-ld: warning: libgstvideo-0.10.so.0, needed by /usr/local/lib/libopencv_videoio.so.3.4.0, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libgstapp-0.10.so.0, needed by /usr/local/lib/libopencv_videoio.so.3.4.0, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libgstriff-0.10.so.0, needed by /usr/local/lib/libopencv_videoio.so.3.4.0, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libgstpbutils-0.10.so.0, needed by /usr/local/lib/libopencv_videoio.so.3.4.0, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libavcodec-ffmpeg.so.56, needed by /usr/local/lib/libopencv_videoio.so.3.4.0, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libavformat-ffmpeg.so.56, needed by /usr/local/lib/libopencv_videoio.so.3.4.0, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libavutil-ffmpeg.so.54, needed by /usr/local/lib/libopencv_videoio.so.3.4.0, not found (try using -rpath or -rpath-link)
/usr/bin/x86_64-linux-gnu-ld: warning: libswscale-ffmpeg.so.3,
```

I googled the missing libraries, saw that ubuntu removed their support after 16.04. It might have been the case that I messed up some other software(maybe opencv).

AKS1996, 2018-05-28:
Sure about the `ffmpeg` package, [refer here](https://askubuntu.com/questions/852605/are-ffmpeg56-and-ffmpeg54-packages-discontinued)

fspindle, 2018-05-28:
Try 

```
sudo apt-get install libswscale-dev libavutil-dev libavformat-dev libavcodec-dev libbz2-dev libbz2-1.0
```

and let me know

AKS1996, 2018-05-28:
I tried. Ubuntu 18.04 has upgraded versions of those libraries(like `libavcodec-ffmpeg57` instead of `libavcodec-ffmpeg56`). `opencv-3.4.0` needs older ones.
 
The work around is to add package `universe` and `multiverse` for 16.04(or older) in 18.04's `/etc/apt/sources.list` and then update. Try not remove newer versions of those libraries(might uninstall some other apps like opencv or vlc)

suuman, 2018-05-29:
To compile visp succesfully with opencv 4 pre version (the one in git hub master)
I have added following:
 
a) In visp/modules/core/include/visp3/core/vpImageConvert.h
after  _#ifdef VISP_HAVE_OPENCV_ (line 58)
**#if (VISP_HAVE_OPENCV_VERSION >= 0x040000)
#include <opencv2/imgproc/types_c.h>
#endif**
to fix _Iplimage_ related error.

b) In visp/modules/vision/include/visp3/vision/vpKeyPoint.h
after line 80
**#if (VISP_HAVE_OPENCV_VERSION >= 0x040000)
#include <opencv2/imgproc/imgproc_c.h>
#endif**
to fix _CV_FILLED_ related error.

fspindle, 2018-05-29:
I'm out of my office until next week. I will check what you mention and introduce a fix ASAP.

fspindle, 2018-06-05:
Compat with OpenCV 4.0.0-pre is fixed in #373.

I checked again; on a fresh Ubuntu 18.04 LTS, ViSP builds with OpenCV installed using `apt-get install libopencv-dev` that brings OpenCV 3.2.0 libraries in `/usr/lib/x86_64-linux-gnu`.

Your issue relates to OpenCV 4.0.0 that you install from github master in `/usr/local` if I'm right. In that case ViSP builds only if some changes are introduced (see #373). Then it builds also without trouble on my 18.04.

The only thing I had to do to execute the binaries was to indicate the location of the OpenCV 4.0.0 libraries using ` export LD_LIBRARY_PATH=/usr/local/lib`

AKS1996, 2018-06-05:
Yes, you're right. That missing libraries issue was solved after the `export LD_LIBRARY_PATH` command. Thanks

closed 2018-06-06 by fspindle