lagadic/visp #380: Compilation error when building the excutable "testRealSense2_D435.cpp"

arrfou90, 2018-06-14:
Hello, I got this compilation error when including  Mutex. I have  GCC 5.4.0, Ubuntu 16.04, RT kernel.

```s/testRealSense_R200.dir/test/rgb-depth/testRealSense_R200.cpp.o
[ 31%] Linking CXX executable testRealSense_R200
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:75:53: error: ‘std::mutex’ has not been declared
   explicit ViewerWorker(const bool color_mode, std::mutex &mutex) : m_colorMode(color_mode), m_mutex(mutex) {}
                                                     ^
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:142:8: error: ‘mutex’ in namespace ‘std’ does not name a type
   std::mutex &m_mutex;
        ^
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp: In constructor ‘{anonymous}::ViewerWorker::ViewerWorker(bool, int&)’:
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:75:94: error: class ‘{anonymous}::ViewerWorker’ does not have any field named ‘m_mutex’
   explicit ViewerWorker(const bool color_mode, std::mutex &mutex) : m_colorMode(color_mode), m_mutex(mutex) {}
                                                                                              ^
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp: In member function ‘void {anonymous}::ViewerWorker::run()’:
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:95:9: error: ‘unique_lock’ is not a member of ‘std’
         std::unique_lock<std::mutex> lock(m_mutex, std::try_to_lock);
         ^
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:95:9: note: suggested alternatives:
In file included from /usr/include/boost/thread/pthread/thread_data.hpp:12:0,
                 from /usr/include/boost/thread/thread_only.hpp:17,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /usr/include/pcl-1.7/pcl/visualization/boost.h:57,
                 from /usr/include/pcl-1.7/pcl/visualization/common/actor_map.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/leo/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:54:
/usr/include/boost/thread/lock_types.hpp:40:9: note:   ‘boost::unique_lock’
   class unique_lock;
         ^
In file included from /usr/include/boost/signals2/connection.hpp:22:0,
                 from /usr/include/boost/signals2/signal.hpp:22,
                 from /usr/include/boost/signals2.hpp:19,
                 from /usr/include/pcl-1.7/pcl/visualization/boost.h:52,
                 from /usr/include/pcl-1.7/pcl/visualization/common/actor_map.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/leo/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:54:
/usr/include/boost/signals2/detail/unique_lock.hpp:24:13: note:   ‘boost::signals2::detail::unique_lock’
       class unique_lock: public noncopyable
             ^
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:95:26: error: ‘mutex’ is not a member of ‘std’
         std::unique_lock<std::mutex> lock(m_mutex, std::try_to_lock);
                          ^
/home/user1/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:95:26: note: suggested alternatives:
In file included from /usr/include/boost/thread/mutex.hpp:16:0,
                 from /usr/include/boost/thread/pthread/thread_data.hpp:13,
                 from /usr/include/boost/thread/thread_only.hpp:17,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from /usr/include/pcl-1.7/pcl/visualization/boost.h:57,
                 from /usr/include/pcl-1.7/pcl/visualization/common/actor_map.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/leo/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:54:
/usr/include/boost/thread/pthread/mutex.hpp:92:11: note:   ‘boost::mutex’
     class mutex
           ^
In file included from /usr/include/boost/signals2/mutex.hpp:30:0,
                 from /usr/include/boost/signals2/signal.hpp:31,
                 from /usr/include/boost/signals2.hpp:19,
                 from /usr/include/pcl-1.7/pcl/visualization/boost.h:52,
                 from /usr/include/pcl-1.7/pcl/visualization/common/actor_map.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/leo/work/visp/modules/sensor/test/rgb-depth/testRealSense2_D435.cpp:54:
/usr/include/boost/signals2/detail/lwm_pthreads.hpp:30:7: note:   ‘boost::signals2::mutex’
 class mutex
```
Any help would be greatly appreciated.
Arrfou

fspindle, 2018-06-14:
Check if in cmake `USE_CPP11` is enabled.

arrfou90, 2018-06-14:
thanks for your reply @fspindle. I have checked and it was enabled even when the gcc version I had does not support c++11. I have upgraded to the gcc version to 6 and it solved the issue.

closed 2018-06-14 by arrfou90