lagadic/visp #418: GSOC'18 Work Submission Report

AKS1996, 2018-08-12:
Hi @s-trinh @fspindle 

I thought about a github gist or wiki to summarize the project. I chose wiki a because you can add it easily in visp wiki's(like [this wiki](https://github.com/lagadic/visp/wiki/GSoC_2018)). You can get the [wiki here](https://github.com/AKS1996/visp/wiki/GSoC18_Work_Submission). 

Note that I'll be providing the link to wiki(after you add it to visp.wiki) to GSoC final evaluation page.

All suggestions are welcome. I thought like this:
+ Whats ViSP?
+ Brief intro to the project
+ Listing and describing the PR's
+ A table listing tasks completed, and future scope
+ Links to tutorials for beginners(to `java_module` and `android_platform`)
+ Listing blogs I wrote, links to all commits, etc.

AKS1996, 2018-08-13:
@s-trinh Thanks for your suggestion. I've added the changes, do have a look at them.

If everything's appropriate, then I'll look forward for getting it merged so that I can complete the evaluation today. 
Meanwhile I'll update the other doc and begin complete `apriltag` detection sample.

s-trinh, 2018-08-13:
Thanks, I have updated the ViSP [Wiki](https://github.com/lagadic/visp/wiki/GSoC-2018:-Porting-ViSP-to-Android-devices).

AKS1996, 2018-08-14:
Thanks. Completed the evaluation just now.

closed 2018-08-14 by AKS1996