lagadic/visp #433: Can not build android SDK

ZipirMario, 2018-09-15:
Hello,
 I am trying to build the android SDK following steps in this "http://visp-doc.inria.fr/doxygen/visp-daily/tutorial-create-android-sdk.html" page. In last step i am taking error.
 Guide says use that "$ python build_sdk.py --config ndk.config.py --sdk_path <path-to-Android-Sdk> --ndk_path <path-to-Android-Sdk>/ndk-bundle <installation-directory> <visp-source-code-directory>"
My commend is "$ sudo python build_sdk.py --config ndk.config.py --sdk_path /Users/mac/Library/Android/sdk --ndk_path /Users/mac/Library/Android/sdk/ndk-bundle /Users/mac/Desktop/visp /Users/mac/Desktop/visp-snapshot-2018-09-14" 
But i am taking this error : "  File "build_sdk.py", line 366, in <module>
    builder.build_library(abi, do_install)
  File "build_sdk.py", line 175, in build_library
    execute(cmd)
  File "build_sdk.py", line 25, in execute
    raise Fail("Execution failed: %d / %s" % (e.errno, e.strerror))
__main__.Fail: Execution failed: 2 / No such file or directory " 

How can i solve this?

AKS1996, 2018-09-16:
Hi @ZipirMario. Can you provide the complete build log, so that we can get more details about why build failed

ZipirMario, 2018-09-16:
It is like this.

Hamza-MacBook-Pro:android hamzaozder$ ls
android.toolchain.cmake	ndk.config.py
build_sdk.py		service
Hamza-MacBook-Pro:android hamzaozder$ sudo python build_sdk.py --config ndk.config.py --sdk_path /Users/hamzaozder/Library/Android/sdk --ndk_path /Users/hamzaozder/Library/Android/sdk/ndk-bundle /Users/hamzaozder/Desktop/visp /Users/hamzaozder/Desktop/visp-snapshot-2018-09-14
Args: Namespace(build_doc=False, config='ndk.config.py', extra_modules_path=None, extra_pack=None, force_copy=False, force_visp_toolchain=False, ndk_path='/Users/hamzaozder/Library/Android/sdk/ndk-bundle', no_ccache=False, sdk_path='/Users/hamzaozder/Library/Android/sdk', sign_with=None, visp_dir='/Users/hamzaozder/Desktop/visp-snapshot-2018-09-14', work_dir='/Users/hamzaozder/Desktop/visp')
Package configuration:
================================================================================
ABIs = [
    ABI("2", "armeabi-v7a", "arm-linux-androideabi-4.9", cmake_vars=dict(ANDROID_ABI='armeabi-v7a with NEON')),
    ABI("3", "arm64-v8a",   "aarch64-linux-android-4.9"),
    ABI("5", "x86_64",      "x86_64-4.9"),
    ABI("4", "x86",         "x86-4.9")
]
================================================================================
Android NDK path: /Users/hamzaozder/Library/Android/sdk/ndk-bundle
Android SDK path: /Users/hamzaozder/Library/Android/sdk
Check dir /Users/hamzaozder/Desktop/visp (create: True, clean: False)
Check dir /Users/hamzaozder/Desktop/visp-snapshot-2018-09-14 (create: False, clean: False)
Check dir /Users/hamzaozder/Desktop/visp/o4a (create: True, clean: True)
Check dir /Users/hamzaozder/Desktop/visp/ViSP-android-sdk (create: True, clean: True)
Removing dir: /Users/hamzaozder/Desktop/visp/ViSP-android-sdk/sdk
Check dir /Users/hamzaozder/Desktop/visp/ViSP-android-sdk/sdk/java/javadoc (create: True, clean: True)
Detected ViSP version: 3.2.0-1
Detected Engine version: 3.40
Check dir /Users/hamzaozder/Desktop/visp/build_service_armeabi-v7a (create: True, clean: True)
=====
===== Building library for armeabi-v7a (arm-linux-androideabi-4.9)
=====
Executing: ['cmake', '-GNinja', '-DNDK_CCACHE=ccache', "-DBUILD_ANDROID_EXAMPLES='ON'", "-DINSTALL_ANDROID_EXAMPLES='ON'", "-DBUILD_EXAMPLES='OFF'", "-DBUILD_TESTS='OFF'", "-DBUILD_DEMOS='OFF'", "-DANDROID_TOOLCHAIN_NAME='arm-linux-androideabi-4.9'", "-DANDROID_STL='gnustl_static'", "-DANDROID_ABI='armeabi-v7a with NEON'", "-DCMAKE_TOOLCHAIN_FILE='/Users/hamzaozder/Library/Android/sdk/ndk-bundle/build/cmake/android.toolchain.cmake'", "-DBUILD_TUTORIALS='OFF'", '/Users/hamzaozder/Desktop/visp-snapshot-2018-09-14']
Executing: cmake -GNinja -DNDK_CCACHE=ccache -DBUILD_ANDROID_EXAMPLES='ON' -DINSTALL_ANDROID_EXAMPLES='ON' -DBUILD_EXAMPLES='OFF' -DBUILD_TESTS='OFF' -DBUILD_DEMOS='OFF' -DANDROID_TOOLCHAIN_NAME='arm-linux-androideabi-4.9' -DANDROID_STL='gnustl_static' -DANDROID_ABI='armeabi-v7a with NEON' -DCMAKE_TOOLCHAIN_FILE='/Users/hamzaozder/Library/Android/sdk/ndk-bundle/build/cmake/android.toolchain.cmake' -DBUILD_TUTORIALS='OFF' /Users/hamzaozder/Desktop/visp-snapshot-2018-09-14
Traceback (most recent call last):
  File "build_sdk.py", line 366, in <module>
    builder.build_library(abi, do_install)
  File "build_sdk.py", line 175, in build_library
    execute(cmd)
  File "build_sdk.py", line 25, in execute
    raise Fail("Execution failed: %d / %s" % (e.errno, e.strerror))
__main__.Fail: Execution failed: 2 / No such file or directory
Hamza-MacBook-Pro:android hamzaozder$

ZipirMario, 2018-09-17:
Also i am taking this Error,

Hamza-MacBook-Pro:android hamzaozder$ ls
android.toolchain.cmake	ndk.config.py
build_sdk.py		service
Hamza-MacBook-Pro:android hamzaozder$ sudo python build_sdk.py --config ndk.config.py --sdk_path /Library/Android/sdk --ndk_path /Library/Android/sdk/ndk-bundle /Users/hamzaozder/Desktop/visp /Users/hamzaozder/Downloads/visp-snapshot-2018-09-17/platforms/android
Args: Namespace(build_doc=False, config='ndk.config.py', extra_modules_path=None, extra_pack=None, force_copy=False, force_visp_toolchain=False, ndk_path='/Library/Android/sdk/ndk-bundle', no_ccache=False, sdk_path='/Library/Android/sdk', sign_with=None, visp_dir='/Users/hamzaozder/Downloads/visp-snapshot-2018-09-17/platforms/android', work_dir='/Users/hamzaozder/Desktop/visp')
Package configuration:
================================================================================
ABIs = [
    ABI("2", "armeabi-v7a", "arm-linux-androideabi-4.9", cmake_vars=dict(ANDROID_ABI='armeabi-v7a with NEON')),
    ABI("3", "arm64-v8a",   "aarch64-linux-android-4.9"),
    ABI("5", "x86_64",      "x86_64-4.9"),
    ABI("4", "x86",         "x86-4.9")
]
================================================================================
Android NDK path: /Library/Android/sdk/ndk-bundle
Android SDK path: /Library/Android/sdk
Check dir /Users/hamzaozder/Desktop/visp (create: True, clean: False)
Check dir /Users/hamzaozder/Downloads/visp-snapshot-2018-09-17/platforms/android (create: False, clean: False)
Check dir /Users/hamzaozder/Desktop/visp/o4a (create: True, clean: True)
Check dir /Users/hamzaozder/Desktop/visp/ViSP-android-sdk (create: True, clean: True)
Removing dir: /Users/hamzaozder/Desktop/visp/ViSP-android-sdk/sdk
Check dir /Users/hamzaozder/Desktop/visp/ViSP-android-sdk/sdk/java/javadoc (create: True, clean: True)
Traceback (most recent call last):
  File "build_sdk.py", line 342, in <module>
    builder = Builder(args.work_dir, args.visp_dir, args)
  File "build_sdk.py", line 130, in __init__
    self.visp_version = determine_visp_version(os.path.join(self.vispdir, "CMakeLists.txt"))
  File "build_sdk.py", line 56, in determine_visp_version
    with open(version_hpp_path, "rt") as f:
IOError: [Errno 2] No such file or directory: '/Users/hamzaozder/Downloads/visp-snapshot-2018-09-17/platforms/android/CMakeLists.txt'
Hamza-MacBook-Pro:android hamzaozder$

fspindle, 2019-01-16:
Fixed in PR #507

closed 2019-01-16 by fspindle