lagadic/visp #474: How to get object position for İOS?

dpyheeau, 2018-12-05:
Hi,
I’m developing an iOS app which includes models based tracking with the help of object detection and localization tutorials. I have no problem with detection and tracking but cannot solve the issue about getting the position of the object for 3d model. In order to specify my issue;

In VISP: 
if (keypoint_detection.matchPoint(I, cam, cMo, error, elapsedTime)) {
        tracker.setPose(I, cMo);
        tracker.display(I, cMo, cam, vpColor::red, 2);
        vpDisplay::displayFrame(I, cMo, cam, 0.025, vpColor::none, 3);
 }

In my opinion, I quite possibly need a line in if condition like;
tracker.get3dModelPosition();

Any help would be great and appreciated.

fspindle, 2018-12-06:
In the code you mention, `keypoint_detection.matchPoint()` has parameter `cMo` that is an output of the function and that contains the position of the object wrt the camera.

Note that I recently introduced [tutorial-mb-generic-tracker-live.cpp](https://github.com/lagadic/visp/blob/master/tutorial/tracking/model-based/generic/tutorial-mb-generic-tracker-live.cpp) that shows how to learn an object (using `--learn` command line option) and then how to use the learned database to automize initial detection using `--auto-init` option. This example sounds similar to what you want to implement on iOS.

dpyheeau, 2018-12-07:
Thank you so much for your quick answer and explanation also great tutorial. I checked and applied the code from the tutorial and all is fine. But I have one important question regarding cMo matrix. 
For example, for the cMo matrix below:
0.3707500255  -0.4094446749  -0.8336063081  -0.1030196426
0.5477073695  0.8212696227  -0.1597900006  -0.1140818129
0.7500407029  -0.3973301714  0.5287415994  0.5201641872
0  0  0  1
I try to understand which value(s) among 12 gives the position of the object because I want to apply a drawing (text, image, model) on the object? Or do I need to check or apply another method for example directly from AR module?

fspindle, 2018-12-07:
The 3D position is given by the last column that you can get using
```
vpTranslationVector cto = cMo.getTranslationVector();
```
The 3D orientation is given by the upper-left 3 by 3 bloc that you can get using
```
vpRotationMatrix cRo = cMo.getRotationMatrix();
```
that could be converted in Quaternion, Euler angles...
```
vpQuaternionVector cqo(cRo);
vpRxyzVector crxyzo(cRo);
```

dpyheeau, 2018-12-07:
I've found the 3d position successfuly with the help of providing cMo and cam parameters to displayFrame method used in imageDisplay class in "AprilTag marker detection on iOS" tutorial. Thank you so much.

closed 2018-12-07 by fspindle

Logeshmuthu, 2024-03-28:
@dpyheeau Hi,

Now i also trying the visp tracker but model not fitting properly 

if you don't mine give me your sample code for visp tracking, it's help for me

Thank You.