lagadic/visp #512: Gaussian blur for color images

yshkp, 2019-01-17:
Tutorial about implementing `vpImageFilter::gaussianBlur()` on RGB color images must be introduced.Extending it may be by introducing simultaneous grabbing image using color camera.

yshkp, 2019-01-20:
@fspindle I would like you to have a view and give a brief so that I can start working on it. Also I would like to hear your suggestions.

fspindle, 2019-01-20:
@yshkp Let's go on

fspindle, 2019-01-22:
@yshkp Step by step. Sart with gaussianBlur() and functions that are called inside.

yshkp, 2019-01-31:
@fspindle Please look into my doubt mentioned in previous comment.Do I need to create a test for it?

yshkp, 2019-01-31:
@fspindle Please look into my doubt mentioned in previous comment.Do I also need to create a test for it?

commit 13624391 (2019-02-28): Fix #512

Now, Gaussian blur is available for RGB color images with all formats specified earlier.
commit f9239bec (2019-02-28): Merge pull request #528 from yshkp/master

Build Gaussian Blur for color images and closes #512
closed 2019-02-28 by fspindle