lagadic/visp #538: Apriltag id not available in Android / Java wrapper

jukindle, 2019-02-27:
The c++-way of reading the tag-id is parsing the message from the detector. There is however no function in the Java wrapper which returns this message. If I am wrong, could somebody tell me how to get the apriltag ids corresponding to the HomogeneousMatrices?

s-trinh, 2019-03-25:
To get the tag-id, I see two options:

- add a method in `vpDetectorAprilTag` that returns a vector of int for the tag-id
- wrap the method to be available in Java [here](https://github.com/lagadic/visp/blob/44e5e1fd8e94e15c1c5bb2e0bc882478f27afcff/modules/java/misc/detection/gen_dict.json)

or use JNI to call C++ code from Java, something similar to [this](https://github.com/AKS1996/GSOC-Prep/blob/cb3f297a55044115eb060f173e1f47db48861ea4/visp-android-demos/VispAndroidDemoApp/app/src/main/cpp/native-lib.cpp)

A pull request will be appreciated if you implement and wrap a method that returns the tag-id for Java.

bigcmos, 2019-03-28:
Do you have a how to get started building the Android app?

fspindle, 2019-03-28:
Do you have seen this [project](https://github.com/AKS1996/GSOC-Prep/tree/master/visp-android-demos/AprilTagDetection) ?

[Here](https://github.com/AKS1996/GSOC-Prep) are also a set of java and android examples.

PR #680 (2020-01-29): Update Apriltag test and perf
needs lagadic/visp-images#11

closes #538 

Remove `const` for native datatype in function parameters, [see](https://abseil.io/tips/109).
Since Apriltag detection should be one of the most used functionality, add a proper test procedure to avoid issue like this: #667
closed 2020-01-29 by fspindle