lagadic/visp #549: Bug in unsigned int vpBSpline::findSpan(double u) ?

romainla, 2019-03-20:
Hello,
There is something weird in the function [findSpan](https://visp-doc.inria.fr/doxygen/visp-daily/classvpBSpline.html#a4446856f29efe34528f9c33a9139ceb2) behaviour. I tested the inputs described in the documentation of the function and did not obtain the outputs that are said to be expected. I created a small CMake project  
[vpBSpline.zip](https://github.com/lagadic/visp/files/2987788/vpBSpline.zip)
to check that. 

Would you mind to tell me if it is because of a typo in the documentation or a bug in the function please ? I'm trying to develop a vpBSplineSurfaces for the ViSP library and wanted to base my work on the existing class.

Regards,
Romain

fspindle, 2019-03-26:
Your code sounds correct to me.
I'm not familiar with BSpline. I cannot figure out if this is an error in the doc or not.
Could you investigate more from the theory ?

fspindle, 2019-03-31:
Fixed in PR #556

closed 2019-03-31 by fspindle