lagadic/visp #559: AprilTag detector crashes under Windows when nthread parameter greater than 1

fspindle, 2019-04-01:
This issue occurs on Windows with msvc14 and msvc15 when `3rdparty/pthreads-win32` is used.

It could be reproduced running:
````
tutorial-apriltag-detector --nthreads 2
````

s-trinh, 2019-04-01:
If I remember correctly, I think that the issue is not present anymore when pthreads is built in shared library.

fspindle, 2019-04-02:
Not checked recently, but from what I remmember you are right. Note that the libraries in 3rdparty folder should be built as static libs.

fspindle, 2019-05-21:
To keep a trace after some tests:
- the freeze doesn't occurs with MinGW that uses native pthread library
- the freeze occurs with msvc when pthread built-in is used.
- To reproduce the issue run:
    ```
    tutorial-apriltag-detector.exe --nthreads 2
    ```
- It hangs in `workerpool.c` [L188](https://github.com/lagadic/visp/blob/master/3rdparty/apriltag/common/workerpool.c#L188) that calls `pthread_cond_wait(&wp->startcond, &wp->mutex)`

PR #583 (2019-05-28): Fix pthreads static build
fix #559

Using [pthreads4w](https://sourceforge.net/projects/pthreads4w/) solves the issue for me.
closed 2019-05-28 by fspindle