lagadic/visp #569: How can I get object position in generic stereo model based tracking?

orz-000, 2019-04-16:
I used generic stereo model based tracking and I want object position(x,y,z,roll,pitch,yaw) for ROS system.
How can I get it? That system have a getPose() function but it doesn't return anything.

fspindle, 2019-04-23:
There exists [vision_visp](http://wiki.ros.org/vision_visp) meta package that contains [visp_tracker](http://wiki.ros.org/visp_tracker) that allows to track an object using a monocular camera. Sorry, but this package was not extented to handle stereo cameras. 

I plan to introduce before the summer a new package `visp_generic_tracker` that will support this feature. In the mean time you have to write your own ROS node.

orz-000, 2019-04-24:
Thank you for replying. I'm trying to write ros pkg and I have some questions.
First question, Why catkin can't find visp repository?
https://github.com/ros/catkin/issues/606
I saw this issue but I can't understand, if you are okey, I want you to explain.
 Second question, you consolidated lib and header files of visp for writing ros pkg, right?
Visp tracker ros pkg have few lib and header files, but origin visp pkg, visp_generic_tracker, have so many lib and header packages and I  should import these numerous files into catkin. What do you think about best practice to write ros pkg of visp_stereo_model_tracker?

closed 2021-02-17 by fspindle