lagadic/visp #570: Could not get left and right Infrared images from RealSense D435

eddylan2894, 2019-04-23:
I am trying to get both the IR images from an RS D435 using the RealSense D435 SDK like this:
```
    config.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);
    config.enable_stream(RS2_STREAM_INFRARED, 2, 640, 480, RS2_FORMAT_Y8, 30);
```
However, it always throws an error when I tried to include the second IR stream:
`
  terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
`
Do you know how I can resolve this issue? Thanks!

fspindle, 2019-04-24:
Should work with:
```
config.enable_stream(RS2_STREAM_INFRARED, 0, 640, 480, RS2_FORMAT_Y8, 30);
config.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);
```

eddylan2894, 2019-04-24:
Issue solved by updating the RealSense's SDK and firmware. Thanks.

closed 2019-04-24 by fspindle