lagadic/visp #573: VISP 3.2.0 fails to build with make 3.14.3 and PCL 1.9.1:

fxcoudert, 2019-05-03:
VISP 3.2.0 fails to build with make 3.14.3 and PCL 1.9.1, on macOS:

```
CMake Error at /usr/local/share/pcl-1.9/PCLConfig.cmake:361 (pcl_report_not_found):
  cmake_policy POP without matching PUSH
Call Stack (most recent call first):
  /usr/local/share/pcl-1.9/PCLConfig.cmake:545 (find_external_library)
  cmake/VISPUtils.cmake:533 (find_package)
  CMakeLists.txt:597 (VP_OPTION)

CMake Error in /usr/local/share/pcl-1.9/PCLConfig.cmake:
  cmake_policy PUSH without matching POP
Call Stack (most recent call first):
  cmake/VISPUtils.cmake:533 (find_package)
  CMakeLists.txt:597 (VP_OPTION)
```

fxcoudert, 2019-05-03:
Not sure whether this is a VISP or PCL bug: https://github.com/PointCloudLibrary/pcl/issues/3062

fspindle, 2019-05-06:
The issue you refer was already fixed in ViSP master branch.
But when using the master branch with pcl-1.9.1 there is an other issue arround c++-14 required by pcl  (see #574).

closed 2019-05-06 by fspindle