lagadic/visp #574: Compatibility problem with PCL1.9.1

romainla, 2019-05-06:
We faced a compilation problem while compiling ViSP with PCL1.9.1 obtained from the Github repository of the Point Cloud Library.

This is due to the fact that PCL1.9.1 requires CMAKE_CXX_FLAGS=-std=c++14  option while ViSP is currently using the c++11 standard.

We found a hack modifying the standard in different CMake files, I created a [patch](https://github.com/lagadic/visp/files/3146791/patchForPCL1_9_1.txt)  we can be used with the command `patch -p1 < patchForPCL1_9_1.txt` 

I did the trick but it is not really pretty though, especially because the variable is called USE_CXX11.

I hope it can be of use for people facing similar problem.

Let me know if you need more information,
Regards

Romain

fspindle, 2019-05-06:
This issue occurs with the master branch where `CMAKE_CXX_STANDARD = 14` was introduced by default.

It doesn't occur with pcl 1.9.1.

To get pcl 1.9.1 from github you should do:
```
$ git clone https://github.com/PointCloudLibrary/pcl.git
$ git checkout pcl-1.9.1
```
Could be checked using:
```
$ git branch
* (HEAD detached at pcl-1.9.1)
  master
```

In your case I'm pretty sure that you have:
```
$ git branch
  master
```
which indicates that you are on the master branch.

fspindle, 2019-05-06:
Issues occuring with pcl master branch look like:
```
/home/user/pcl/build/install/include/pcl-1.9/pcl/pcl_config.h:7:4: error: #error PCL requires C++14 or above
   #error PCL requires C++14 or above
```

romainla, 2019-05-09:
I just checked you are right it seems that I cloned the master branch instead of the pcl-1.9.1 branch as I intended to do. Sorry for this mistake.

fspindle, 2019-05-10:
Fixed in PR #577

closed 2019-05-10 by fspindle