lagadic/visp #575: REALSENSE2_DIR and PCL_ROOT environment variables

romainla, 2019-05-06:
Hello,

The REALSENSE2_DIR and PCL_ROOT variables used for the compilation of ViSP are assumed to be environment variables and cannot be given through CMake commands. It may be problematic when you want to compile ViSP with a version of one of these libraries that is not installed on your machine (e.g. you want to give a try to the current dev version of PCL so you compile it but don't want to install it on your computer).

Do you think it could be changed to make the ViSP building process a bit more modular ?

Thank you for your help,
Regards

Romain

fspindle, 2019-05-10:
Can you check again. I'm not able to reproduce what you mention.

First unset existing env var:
```
$ echo $PCL_ROOT
<path_pcl>/pcl/build/install/share/pcl-1.9
$ echo $REALSENSE2_DIR
<path_librealsense>/librealsense/build/install
$ unset PCL_ROOT
$ unset REALSENSE2_DIR
```
Then configure ViSP to verify that pcl and realsense are not found
```
$ cd visp-build
$ cmake ../visp
$ grep PCL ViSP-third-party.txt 
    Use PCL:                     no
$ grep Realsense2 ViSP-third-party.txt 
    Use Realsense2:              no
```
Then configure with the appropriate path using `-DPCL_ROOT` and `-DREALSENSE2_DIR`:
```
$ cmake ../visp -DPCL_ROOT=<path_pcl>/pcl/build/install/share/pcl-1.9 -DREALSENSE2_DIR=<path_librealsense>/librealsense/build/install
```
Here you should have something similar to:
```
$ cd visp-build
$ cmake ../visp
$ grep PCL ViSP-third-party.txt 
    Use PCL:                     yes (ver 1.9.1.99)
$ grep Realsense2 ViSP-third-party.txt 
    Use Realsense2:              yes (ver 2.19.1)
```

romainla, 2019-05-10:
I don't understand I did the manipulation you asked me to do and it works. Being unable to

closed 2019-05-10 by romainla