lagadic/visp #580: Model based tracking without user intervention

holo559, 2019-05-21:
How can I track a 3D model without giving 4 initial points. My model must be tracked automatically. There should not be any user intervention. How can I achieve this using ViSP?

fspindle, 2019-05-21:
You can explore this [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-object.html) that shows how to learn your object using keypoints.  

Depending on your application, you can also glue a tag (see [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-apriltag.html)) on your object. It's pose could be used to initialize the tracker (see [code example](https://github.com/lagadic/visp/tree/master/tutorial/tracking/model-based/generic-apriltag)) and [video](https://www.youtube.com/watch?v=pr27-OZEzPs).

closed 2019-05-21 by fspindle

holo559, 2019-05-21:
@fspindle I want to track model but I don't want to initialize as described in this [tutorial](https://visp-doc.inria.fr/doxygen/visp-daily/tutorial-detection-object.html). I want to achieve something like [this](https://www.youtube.com/watch?v=GfO6hmk7kww). Is there any method which I can use to auto initialize my 3D model instead of providing min. 4 points to track the object? I do not want any kind of human interaction in tracking.

fspindle, 2019-05-21:
Just to be clear with the tutorial I mention. There are 2 stages: 

1. a learning stage that requires manual initialisation on at least one learning image, 
2. an auto initialization stage that doesn't require user interaction

Appart this, there is no other algorithm in ViSP that allows to auto initialize the model-based tracker.