lagadic/visp #582: Generation of .init file

holo559, 2019-05-23:
How can I generate .init file for my CAD model? Is there any plug-in for that? If there is any tutorial please share the link.

fspindle, 2019-05-23:
Usually, we copy/paste coordinates of 3D points from .cao file.

There is no dedicated plug-in or tool.

holo559, 2019-06-10:
@fspindle How should I select 4 points to track the complex 3d model? On what basis you chose those 4 points in [lego bricks tracking](https://www.youtube.com/watch?v=zD2nrl5yqxA).

fspindle, 2019-09-02:
My answer is maybe to late...

To select which 3D points to put in the .init file that are used to initialize manually the tracker, you should ensure that:
- The projection of the points (usually we use four 3D points but could be more in the .init file) in the image must be visible
- The spatial distribution of the projection of points in the image should be as wide as possible in the image (ie they should not be distributed over a very small part of the image) and they should be not aligned

closed 2021-02-17 by fspindle