lagadic/visp #586: Using ViSP for Microsoft HoloLens

holo559, 2019-06-03:
Have you tested ViSP library for detecting and tracking model targets on Microsoft Hololens? If yes, then can you share the tutorial for that. And if no, can you make one for the same?

s-trinh, 2019-06-05:
What are these "model targets on Microsoft Hololens"?

holo559, 2019-06-06:
@s-trinh Model targets are simply real world 3D objects. e.g. Toys, Industrial Equipments, Vehicles etc. Check this [link](https://www.youtube.com/watch?v=6W7_ZssUTDQ) to explore more. Have a look at this [link](https://www.youtube.com/watch?v=GfO6hmk7kww) for model tracking.

fspindle, 2019-06-06:
@holo559 Hololens + model-based tracking was never tested on our side. Since we don't have an hololens in the team, we are unable to share a tutorial. It could be a great opportunity for you to contribute to ViSP...

closed 2019-09-02 by fspindle