lagadic/visp #589: Error when using vpDisplayX and std_msgs::Bool (ROS)

francofusco, 2019-06-07:
Hello,
I was writing a ROS node that uses ViSP (v 3.1) for object detection. I run into a strange (at least to me) conflict between Bool ROS messages and the vpDisplayX header. I really don't know why it happens and I do not know if the source of problems is either the ROS message header or DisplayX, but I think it is at least worth reporting it.

When both headers are included, if `std_msgs::Bool` objects are used then the compiler issues various errors, *e.g.*,
```
expected unqualified-id before ‘int’
std_msgs::Bool msg;
          ^
```

A minimal example that gives this error on my machine is:
```c++
#include <std_msgs/Bool.h>
#include <visp3/gui/vpDisplayX.h>

int main() {
  std_msgs::Bool msg;
  return 0;
}
```
and the CMakeLists file I used is just as follows:
```cmake
cmake_minimum_required(VERSION 2.8)
project(bool_bug)

find_package(catkin REQUIRED COMPONENTS std_msgs)
find_package(VISP REQUIRED)

catkin_package()

include_directories(${catkin_INCLUDE_DIRS} ${VISP_INCLUDE_DIRS})

add_executable(bool_error src/bool_error.cpp)
target_link_libraries(bool_error ${catkin_LIBRARIES} ${VISP_LIBRARIES})
```

As soon as I comment out the `#include <visp3/gui/vpDisplayX.h>`, compiler errors disappear. Interestingly, other messages such as `Int32` seem to work fine instead - but I haven't tested all of them.

Could anyone try to reproduce this? And do you have any idea why this happens? If you think this should be reported to ROS as well, I'll open an issue there.

Finally, if anyone encounters the same issue, a possible workaround is to use the vpDisplayOpenCV instead.

fspindle, 2019-06-07:
Try inverting the 2 following lines:

```
#include <std_msgs/Bool.h>
#include <visp3/gui/vpDisplayX.h>
```

francofusco, 2019-06-07:
Doing it gives the message
```
In file included from /usr/include/x86_64-linux-gnu/visp3/gui/vpDisplayX.h:48:0
/opt/ros/melodic/include/std_msgs/Bool.h:48:50: error: multiple types in one declaration
 typedef ::std_msgs::Bool_<std::allocator<void> > Bool;
```
followed by other errors. I did not notice it before, but all errors are basically due to line 48 of vpDisplayX.h, which is `#include <X11/Xlib.h>`. I checked that header and found the problem: `#define Bool int`.

If I manually un-define it after including vpDisplayX I can compile the code, even though I am not sure this is the best solution.

In any case, thanks for the hint!

closed 2019-06-07 by francofusco